kwindrem/SetupHelper #1: Setup helper added 6x relays front page

RobDuthie, 2021-09-02:
//////// modified order to put Settings, then Notifications at top of list. Added in page switching for 6x relays.

import QtQuick 1.1
import "utils.js" as Utils
import com.victron.velib 1.0

MbPage {
	id: root
	title: qsTr("Device List")
    property VBusItem moveSettings: VBusItem { id: moveSettings; bind: Utils.path("com.victronenergy.settings", "/Settings/GuiMods/MoveSettings")}
    property bool settingsAtTop: moveSettings.valid && moveSettings.value === 1

	model: VisualModels {
//////// put Settings at top of list
        VisualItemModel {
            MbSubMenu {
                description: qsTr("Settings")
                subpage: Component { PageSettings {} }
                show: settingsAtTop
            }

//////// put Notifications second
            MbSubMenu {
                id: menuNotificationsTop
                description: qsTr("Notifications")
                item: VBusItem {
                    property variant active: NotificationCenter.notifications.filter(
                                                 function isActive(obj) { return obj.active} )
                    value: active.length > 0 ? active.length : ""
                }
                subpage: Component { PageNotifications {} }
                show: settingsAtTop
		}

		MbSubMenu {
			description: qsTr("Switching")
			subpage: Component { PageSwitching {} }
		}	

	    
            MbOK {
                description: qsTr("Remove disconnected devices")
                value: qsTr("Press to remove")
                show: settingsAtTop && deviceList.disconnectedDevices != 0
                editable: true

                function clicked() {
                    listview.decrementCurrentIndex()
                    deviceList.removeDisconnected()
                }
            }
        }
		VisualDataModel {
			model: VeSortFilterProxyModel {
				model: DeviceList {
					id: deviceList
					onRowsAboutToBeRemoved: {
						for (var i = first; i <= last; i++)
							deviceList.page(i).destroy()
					}
				}
				sortRole: DeviceList.DescriptionRole
				dynamicSortFilter: true
				naturalSort: true
				sortCaseSensitivity: Qt.CaseInsensitive
			}

			delegate: MbDevice {
				iconId: "icon-toolbar-enter"
				service: model.page.service
				subpage: model.page
			}
		}

    VisualItemModel {
            MbSubMenu {
                id: menuNotifications
                description: qsTr("Notifications")
                item: VBusItem {
                    property variant active: NotificationCenter.notifications.filter(
                                                 function isActive(obj) { return obj.active} )
                    value: active.length > 0 ? active.length : ""
                }
                subpage: Component { PageNotifications {} }
                show: !settingsAtTop
            }

            MbSubMenu {
                description: qsTr("Settings")
                subpage: Component { PageSettings {} }
                show: !settingsAtTop
            }

            MbOK {
                description: qsTr("Remove disconnected devices")
                value: qsTr("Press to remove")
                show: !settingsAtTop && deviceList.disconnectedDevices != 0
                editable: true

                function clicked() {
                    listview.decrementCurrentIndex()
                    deviceList.removeDisconnected()
                }
            }
        }
    }

	Component {
		id: vebusPage
		PageVebus {}
	}

	Component {
		id: batteryPage
		PageBattery {}
	}

	Component {
		id: solarChargerPage
		PageSolarCharger {}
	}

	Component {
		id: acInPage
		PageAcIn {}
	}

	Component {
		id: acChargerPage
		PageAcCharger {}
	}

	Component {
		id: tankPage
		PageTankSensor {}
	}

	Component {
		id: motorDrivePage
		PageMotorDrive {}
	}

	Component {
		id: inverterPage
		PageInverter {}
	}

	Component {
		id: pulseCounterPage
		PagePulseCounter {}
	}

	Component {
		id: digitalInputPage
		PageDigitalInput {}
	}

	Component {
		id: temperatureSensorPage
		PageTemperatureSensor {}
	}

	Component {
		id: unsupportedDevicePage
		PageUnsupportedDevice {}
	}

	Component {
		id: meteoDevicePage
		PageMeteo {}
	}

	Component {
		id: evChargerPage
		PageEvCharger {}
	}

    Component {
        id: dcMeterPage
        PageDcMeter {}
    }

	function addService(service)
	{
		var name = service.name

		var page
		switch(service.type)
		{
		case DBusService.DBUS_SERVICE_MULTI:
			page = vebusPage
			break;
		case DBusService.DBUS_SERVICE_BATTERY:
			page = batteryPage
			break;
		case DBusService.DBUS_SERVICE_SOLAR_CHARGER:
			page = solarChargerPage
			break;
		case DBusService.DBUS_SERVICE_PV_INVERTER:
			page = acInPage
			break;
		case DBusService.DBUS_SERVICE_AC_CHARGER:
			page = acChargerPage
			break;
		case DBusService.DBUS_SERVICE_TANK:
			page = tankPage
			break;
		case DBusService.DBUS_SERVICE_GRIDMETER:
			page = acInPage
			break
		case DBusService.DBUS_SERVICE_GENSET:
			page = acInPage
			break
		case DBusService.DBUS_SERVICE_MOTOR_DRIVE:
			page = motorDrivePage
			break
		case DBusService.DBUS_SERVICE_INVERTER:
			page = inverterPage
			break;
		case DBusService.DBUS_SERVICE_TEMPERATURE_SENSOR:
			page = temperatureSensorPage
			break;
		case DBusService.DBUS_SERVICE_SYSTEM_CALC:
			return;
		case DBusService.DBUS_SERVICE_DIGITAL_INPUT:
			page = digitalInputPage
			break;
		case DBusService.DBUS_SERVICE_PULSE_COUNTER:
			page = pulseCounterPage
			break;
		case DBusService.DBUS_SERVICE_UNSUPPORTED:
			page = unsupportedDevicePage
			break;
		case DBusService.DBUS_SERVICE_METEO:
			page = meteoDevicePage
			break;
		case DBusService.DBUS_SERVICE_VECAN:
			return;
		case DBusService.DBUS_SERVICE_EVCHARGER:
			page = evChargerPage
			break
		case DBusService.DBUS_SERVICE_HUB4:
			return;
        case DBusService.DBUS_SERVICE_DCMETER:
            page = dcMeterPage
            break
		default:
			console.log("unknown service " + name)
			return;
		}

		deviceList.append(service, page.createObject(root, {service: service, bindPrefix: service.name}))
		initListView()
	}

	Component.onCompleted: {
		for (var i = 0; i < DBusServices.count; i++)
			addService(DBusServices.at(i))
	}

	Connections {
		target: DBusServices
		onDbusServiceFound: addService(service)
	}
}

```

RobDuthie, 2021-09-02:
For raspi3B+ with 7inch touch screen.
To make the 6x relays active, needs these files. installed.
1: gpio_list
2: PageMain.qml
3: PageSettingsRelay.qml
4: PageSwitching.qml

RobDuthie, 2021-09-02:
Raspi 3B+ VE direct ports issues with serial expansion HAT. VE Direct USB all OK
Not working on these versions of venus:
Testing on ver 2.73 , 2.80.14

kwindrem, 2021-09-02:
These issues are not related to SetupHelper.

The relay issue is really part of RpiGpioSetup

The VE direct issues are not at all related to any of my packages

The CANbus issues are really part of VeCanSetup if they are issues.

closed 2021-09-02 by kwindrem

RobDuthie, 2021-09-02:
> These issues are not related to SetupHelper. Works fine
> 
> The relay issue is really part of RpiGpioSetup Works OK
> 
> The VE direct issues are not at all related to any of my packages Still needs testing for the serial porting
> 
> The CANbus issues are really part of VeCanSetup if they are issues. ? can't Tx data RX OK for USB or pican.
Hi Kevin
Not meant to be issues with your software (sorry for that) your software work very well, impressed. These are reports of what i have found with testing the new release with a raspi 3B+ after version 2.66.
Note: with your software you can now have 6x relay outputs showing on the device page.
If you want to work on this together to get the rest going will make this a full feature device for everybody. I have all the hardware in place i have a Cerbo, Multiplus2, MPPT, smart shunts, raspi 3 and 4 (4 ver1.4 PCB still not working on boot so can't test your software yet. Please advise.
Regards
Rob D
NZ

RobDuthie, 2021-09-16:
Hi Kevin
Re canbus PiCan file to make it more stable i added the spi Frequency command below: Can you update your VeCanSetup file.
I had issues with data comming in the number were all over the place, by adding in the spi frequency the numbers are rock solid now.

#### begin CAN overlay DEV
# PiCAN Hat
# replace DEV with e.g., can0
[all]
dtparam=spi=on
dtoverlay=mcp2515-DEV-overlay,oscillator=16000000,interrupt=25,spimaxfrequency=500000
dtoverlay=spi-bcm2835-overlay
#### end CAN overlay

kwindrem, 2021-09-16:
Done

RobDuthie, 2021-09-16:
Cheers, That made all the difference to stable readings to BMS connected batteries on Canbus.
Running on Ver2.80.17 venus with your VeCanSetup all Ok no issues to date.
Re the raspi relay control and analog inputs, i have done a simple control code change in the PageSettingsRelay.qml page this acivate  4x more relay outputs without any other code change required. Changed .qml to PDF to upload here.
So just a matter of getting these to there own page now?
I have also the analog page working now with the 2 temps and 3 tanks all showing correct data on screen and 
VE Direct USB working OK
VE Direct I2C port still working on that one with configuration etc. It was OK on Ver <2.66.
Canbus USB OK
Canbus SPI port OK
Everything else is OK and working

.
[PageSettingsRelay.qml.pdf](https://github.com/kwindrem/SetupHelper/files/7180929/PageSettingsRelay.qml.pdf)

kwindrem, 2021-09-16:
Couldn't open your PageSettingsRelay.qml file. Last time I looked, the necessary changes were pretty obvious.

Please continue discussions about CANbus in the VeCanSetup repo and about GPIOs in the RpiGpioSetup repo to keep things together. Posting here might cause me to miss something important.