kwindrem/SetupHelper #5: Can't install SetupHelper

CzarofAK, 2021-11-21:
i tried to setup a new Venus device with the latest Venus.
When i run your commands, the following happened:
'''
root@raspberrypi2:~# wget -qO - https://github.com/kwindrem/SetupHelper/archive/current.tar.gz | tar -xzf - -C /data
root@raspberrypi2:~# rm -rf /data/SetupHelper
root@raspberrypi2:~# mv /data/SetupHelper-current /data/SetupHelper
root@raspberrypi2:~# /data/SetupHelper/setup
creating root setup options directory /data/setupOptions
creating package options directory to /data/setupOptions/SetupHelper
creating file set for v2.80~21-large-23
ERROR v2.80~21-large-23 PageSettings.qml no replacement file
ERROR: incomplete file set for v2.80~21-large-23 - can't continue
root@raspberrypi2:~#
'''

CzarofAK, 2021-11-21:
it works with V2.80~29

kwindrem, 2021-11-21:
This means that that version isn't supported in the current build. I build a set of files for previous versions but don't keep all the development build sets. GuiMods can take a lot of time checking previous versions so I try to keep the number of file sets as low as possible.

Glad you got things working with the latest development release.

CzarofAK, 2021-11-21:
Yes, simple way is to install ist on the latest normal venus, set everything up and then upgrade it to the large version.
Looking foward to have the Large as a Stable release.

closed 2021-11-21 by CzarofAK