kwindrem/SetupHelper #6: ERROR: v2.80~33-large-24 PageSettings.qml no replacement file

MichelleWhy, 2021-12-20:
I'm ending up with below error up on new installation.

root@raspberrypi4:~# /data/SetupHelper/setup
creating package options directory to /data/setupOptions/SetupHelper
creating file set for v2.80~33-large-24
ERROR v2.80~33-large-24 PageSettings.qml no replacement file
ERROR: incomplete file set for v2.80~33-large-24 - can't continue

kwindrem, 2021-12-20:
Thanks for letting me know. I need to update SetupHelper to add a file set for ~33-large. Stay tuned.

kwindrem, 2021-12-21:
SetupHelper v3.14 is now available that provides support for v2.80~33-large-24

MichelleWhy, 2021-12-21:
Thanks. It installs ok now and is working.

Minor issue (from console output during install):
svc: warning: unable to control /service/SetupHelper: file does not exist
svc: warning: unable to control /service/SetupHelper: file does not exist

kwindrem, 2021-12-21:
Victron changed the service to a file system overlay, complicating the task of updating a running Service. At some point, the svc commands no longer control the service (down/up/restart). I probably should be rebooting after I do any service changes, but I'm just trying to restart it.

If you see this or don't see an update to, in this case, PackageManager, then just reboot.

closed 2021-12-22 by MichelleWhy