kwindrem/SetupHelper #7: rpi3+ Large versions (through .swu) issues reinstalling SetupHelper and the other apps [GuiTools, VeCanSetup, et al)

virtuvas, 2021-12-27:
Kevin,

I'm very confused with the setup/reinstall procedure I have to follow everytime I want to install/test a new 2.80Large version on my rpi.

tbh, I usually am reasonably OK with all that, but in this case I've lost the plot completely.
So, install .swu file, boots up, once I sort out root passwd I want to install your tools and then connect to the N2K network and go from there.
SetupHelper finds old setup, menu is slightly confusing, eventually get there. 
Then the others gui/vecan/display/etc are not there, although it states that they are/or rather finds them and lists them in the terminal menu.

Having done it twice the last week, and about to upgrade the rpi4 to latest large, it seems that I'm missing something in the logic/steps I have to follow. So what I'm saying is that I fear that having spent a good few months playing with all these configs I should be able to find my way easily. I wonder how newcomers step through the process. Tool(s) ARE brilliant, wonder if docs should have a shorter more concise version like:
A. new install follow blah blah blah
B. new ver using update blah blah
C. new ver using .swu blah this that.

Not sure I make much sense and wonder if I'm alone in being confused each time I do an upgrade on.

cheers

V.

virtuvas, 2021-12-27:
Just to add, I feel that it needs to be clarified that on reinstall you have to go through each and every tool and install it from terminal, looks like SetupHelper is not able to do so. 
Fe, couldn't get brightness and screen blanking working even though I eventually got (from the gui) all components installed and reported as installed. Had to manually create:
/etc/venus/backlight_device and /etc/venus/blank_display_device (and add the right text in them) as well as run 
dbus -y com.victronenergy.settings /Settings/Gui/AutoBrightness SetValue 0
which you've mentioned that it's included in the installation script.

cheers
V.

kwindrem, 2021-12-27:
It should not be that difficult.

Upgrading to a new Venus version should reinstall all packages that were installed prior to the upgrade. I'm not sure why this is not happening on your systems.

On a new system, the easiest way to get started is the SetupHelper blind install procedure. Then you need to get the other packages. You should be able to use the command line running /data/SetupHelper/setup and download and install packages from it's menu system either from SD/USB media or from the internet (GigHub).

I'm working on a complete rewrite to SetupHelper install that uses the GUI menu system for all actions that will make things much easier.

virtuvas, 2021-12-27:
thanks,

definitely not reinstalling anything. But getting the new Large version in .swu format on the root of a usb stick is considered an upgrade? If so, something's wrong on both my rpis for sure.
Will upgrade the 4 later on tonight, I'll try to note down the steps and failures and post here if you wish?
V.

kwindrem, 2021-12-27:
Yes any Venus firmware update using .swu is an "upgrade".

If you want to poke around, the auto-reinstall is handled by /data/rcS.local which calls /data/SetupHelper/reinstallMods. reinstallMods looks in /data/reinstallScriptsList for packages to reinstall. It calls the package setup script to actually do the reinstall.

The log for SetupHelper and all the setup scripts is may shed some light on what is going wrong.

tail -40 /data/log/SetupHelper | tai64nlocal

to give you the last 40 lines of the log file with the time stamp in human readable form

You should be on SetupHelper v3.14. v4.0 is in beta at the moment if you are insisted in something totally different. Check out the ReadMe and screen shots on the GitHub beta branch. I'm currently redoing much of that to make it more responsive but v4.0~17 is available on the beta branch and is working. Not suggesting you take on another unknown however.

By all means post any issues you come across.

virtuvas, 2022-01-03:
good morning and happy new year :-)

too busy not working last week, back to work mode and had a look again before upgrading the rpi4:

I do have rcS.local with:
```
#!/bin/bash

# SetupHelper reinstall all Venus mods
nohup /data/SetupHelper/reinstallMods > /dev/null &

```
but I also have rc.local which runs my ZeroTier One daemon:

```
#!/bin/sh -e
#
# rc.local
#
/usr/sbin/zerotier-one

exit 0
```
Not quite sure if this combo is OK, or should be combined in one or whatever.
Could that be the culprit? 

cheers
V.

kwindrem, 2022-01-03:
There shouldn't be any conflicts between ZeroTier and SetupHelper. rcS.local and rc.local are completely independent of each other.

SetupHelper will add itself to rcS.local if one exists so even that shouldn't be a problem

virtuvas, 2022-01-05:
hi again,

updated the rpi4, your tools didn't install.
Now considering that same SD started with 2.73 and has gone through a series of updates/tests/errors/etc I thought I'd clear /data as much as I can and try again sort of fresh.

so with just:
/conf
/db
/etc
/home
/keys
/log
/lost+found
/themes
/var
/venus
/vrmfilescache
and rcS.local
I started again first trying to installed with the usb stick trick the beta venus-data.tgz
Installed the manager OK after the two reboots, but wouldn't finish installing anything, was just sitting there in the installing... OK line (last line on the gui). Didn't ssh and try manually.

Anyway, since this meant that OS Large features Settings menu disappeared, thought it was related to the beta, so uninstalled it again, cleared the /data folder and started again with the main  venus-data.tgz (the 45kb one) Two reboots got me nowhere, so command line and installed the following:

```
root@raspberrypi4:/data/SetupHelper# ./setup

This package provides support functions and utilities for Venus modification packages
Packages are automatically reinstalled following a Venus OS update
Packages may also be automatically updated from GitHub
  or a USB stick
Previously uninstalled packages can also be installed and configured
  as an option from the menu either from GitHub or from a USB stick

If internet access is not available, you can manually update/install from a USB stick or SD card

Available actions:
  Install and activate (i)
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

  Enable/disable automatic GitHub package updates (g)
  Manually install packages from GitHub or USB stick (p)

Choose an action from the list above: i
installing default packge list
restarting SetupHelper service
restarting GUI
adding SetupHelper/setup to reinstallScriptsList
completed

root@raspberrypi4:/data/SetupHelper# ./packageInstaller

This script allows Venus modification packages to be installed
Previously installed packages are updated automatically, not from this script
If updating from USB stick, insert it now before proceeding

Select the package source GitHub (g) or USB (u): g
SetupHelper is up to date
GuiMods not yet installed - proceeding
WARNING: no version for GuiMods current installation - proceeding
GuiMods needs update but will be updated automatically - skipping
ShutdownMonitor not yet installed - proceeding
WARNING: no version for ShutdownMonitor current installation - proceeding
ShutdownMonitor needs update but will be updated automatically - skipping
RpiDisplaySetup not yet installed - proceeding
WARNING: no version for RpiDisplaySetup current installation - proceeding
RpiDisplaySetup needs update but will be updated automatically - skipping
RpiGpioSetup not yet installed - proceeding
WARNING: no version for RpiGpioSetup current installation - proceeding
[... 53 lines cut ...]
activating GPIO DT overlay
adding RpiGpioSetup/setup to reinstallScriptsList
Reboot system now (y) or issue a reboot manually later (n): n
system must be rebooted to finish installation and activate components
root@raspberrypi4:/data/RpiGpioSetup#
```

I now have:
SetupHelper v3.14
GuiMods v5.5
RpiGpioSetup v3.4
installed and functioning.
VeCanBus is not yet installed, but was before and obvs the effects are on (config.txt and a few other files that were edited then) so CAN bus BMS works just fine.
Rebooted a few times, all's fine.

HOWEVER, OS Large features submenu is missing from the Settings menu list. See:

https://community.victronenergy.com/questions/114259/venus-os-large-features-menu-missing-on-28033large.html?childToView=114273#answer-114273

Seems that will all your work developing, some of us that are trying to follow from early on, should do a clean uninstall of all your features every now and again and start afresh. Still the OS Large issue to sort, guess will be something simple.
I know life would be easier if I just installed it in a new SD card but there are other things installed and messed about and don't want to go through the whole process again.

cheers

V.

kwindrem, 2022-01-05:
I'm fixing the Large features not appearing in Settings now. It's an issue in both the released and beta versions. Sorry.

virtuvas, 2022-01-06:
thanks, that's fixed now, any ideas on the inability to install using the venus-data.tgz ?

kwindrem, 2022-01-06:
I just did a blind install on my Raspberry PI 4 and and a CCGX. Both worked as expected.

I uninstalled SetupHelper, removed it's directory in /data then did the blind install procedure:
rebooting twice with venus-data.tar.gz on a USB stick plugged into the GX device.

I used venus-data.tar.gz from the SetupHelper beta branch (v4.0~26) for this test.

Note that I removed GuiMods from the SetupHelper v4.0 blind install because there's a better, more generic way.

You can put the <packageName>.tar.gz file on the same media as the blind install and once PackageManager is running, it will transfer them. If auto install is enabled, they will also be installed

virtuvas, 2022-01-07:
one rpi works fine, the one on the boat works fine but your menus look slightly off:

[image]

and then got first two records empty on the details screen:

[image]

data looks like this:

```
root@raspberrypi2:/data# ls -la
drwxr-xr-x   25 root     root          4096 Jan  7 15:04 .
drwxr-xr-x   21 root     root          4096 Dec 24 17:15 ..
-rwxr-xr-x    1 501      staff          210 Dec 24 01:57 ._SetupHelper
-rwxr-xr-x    1 501      staff          276 Dec 22 18:03 ._rcS.local
drwxrwxr-x    3 root     root          4096 Dec 25 16:56 GeneratorConnector
drwxrwxr-x    3 root     root          4096 Dec 18 05:33 GuiMods
drwxrwxr-x    3 root     root          4096 Dec 25 16:57 GuiMods2
drwxrwxr-x    3 root     root          4096 Dec 25 18:18 RpiDisplaySetup
drwxrwxr-x    3 root     root          4096 Dec 16 02:05 RpiGpioSetup
drwxrwxr-x    4 root     root          4096 Jan  5 16:47 SetupHelper
drwxrwxr-x    5 root     root          4096 Dec 16 02:30 ShutdownMonitor
drwxrwxr-x    5 root     root          4096 Dec 24 23:02 TankRepeater
drwxrwxr-x    3 root     root          4096 Jan  4 09:30 VeCanSetup
drwxrwxr-x   18 root     root          4096 Jan  6 13:11 ZeroTierOne-1.8.4
drwxr-xr-x    4 root     root          4096 Jan  7 17:26 conf
drwxr-xr-x    2 www-data root          4096 Jan  7 15:10 db
drwxr-xr-x    2 root     root          4096 Feb 16  2021 etc
drwxr-xr-x    4 root     root          4096 Jan  1  1970 home
drwx------    2 root     root          4096 Feb 16  2021 keys
drwxr-xr-x   53 root     root          4096 Jan  7 16:06 log
drwx------    2 root     root         16384 Jan  1  1970 lost+found
-rw-r--r--    1 root     root           620 Jan  7 15:04 packageList
-rw-r--r--    1 root     root           101 Jan  7 15:10 packageVersions
-rwxr-xr-x    1 root     root           330 Dec 24 23:08 rc.local
-rwxr-xr-x    1 root     root           104 Dec 24 23:00 rcS.local
-rw-r--r--    1 root     root           208 Dec 25 18:18 reinstallScriptsList
drwxr-xr-x   11 root     root          4096 Dec 24 23:08 setupOptions
drwxrwxr-x    3 root     root          4096 Dec 25 14:06 smartbms-venus
drwxr-xr-x    3 root     root          4096 Jan  1  1970 themes
drwxr-xr-x    3 root     root          4096 Jan  1  1970 var
drwxr-xr-x    2 root     root          4096 Jan  1  1970 venus
drwxr-xr-x    2 root     root          4096 Aug 17 14:36 vrmfilescache
```
packageVersions:

```
# AUTO GENERATED DO NOT MODIFY
SetupHelper v3.15
GuiMods v5.5
RpiDisplaySetup v3.4
RpiGpioSetup v3.4
```
and packageList:
```
# list of packages managed by SetupHelper
# lines beginning with # are ignored and can be used
#       to remove a package from auto and manual updates
#       or as comments
# blank lines are ignored
# incomplete lines are ignored
# SetupHelper should be first (and not commented out)
#   so it updates resources used by other packages prior to their update
# packages should add themselves to this list during setup

# Package           GitHubUser  Version
SetupHelper         kwindrem    current
GuiMods             kwindrem    current
RpiDisplaySetup     kwindrem    current
[... 19 lines cut ...]
@4000000061d6e5fc0f885374 SetupHelper/packageAutoUpdater: continuing in cleanup script - SetupHelper exiting
@4000000061d6e5fe1854f354 SetupHelper/packageAutoUpdaterCleanup: running SetupHelper setup script to reinstall
@4000000061d6e5fe1c09b23c SetupHelper/packageAutoUpdaterCleanup: reinstalling SetupHelper
@4000000061d6e5fe3b6a78dc SetupHelper/setup: restarting SetupHelper service
@4000000061d6e5ff0578eac4 SetupHelper/setup: restarting GUI
@4000000061d6e5ff0c52694c SetupHelper/setup: completed
@4000000061d6e6011bfca664 SetupHelper/packageAutoUpdater: starting up
@4000000061d6e60125975c14 SetupHelper/updatePackageVersions: creating dbus Setting /Settings/PackageVersion/0/PackageVersion
@4000000061d6eac13b36d234 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d6eac50062c334 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d6eac63ae0b354 SetupHelper/packageAutoUpdater: waiting for dBus settings
@4000000061d6eac730be3fa4 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d6eac8358d7324 SetupHelper/packageAutoUpdater: starting up
@4000000061d838d1223cdea4 SetupHelper/packageAutoUpdaterCleanup: running SetupHelper setup script to reinstall
@4000000061d838d122af99bc SetupHelper/packageAutoUpdaterCleanup: /data/SetupHelper- is missing - can't continue
@4000000061d838e8397334ec SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d838eb3727e55c SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d838ed2ba082ac SetupHelper/packageAutoUpdater: waiting for dBus settings
@4000000061d838ee2d88e604 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061d838ef2991ea14 SetupHelper/packageAutoUpdater: starting up
```

cheers

V.

virtuvas, 2022-02-10:
everything works fine, just realised this is still open, so closing!

closed 2022-02-10 by virtuvas