kwindrem/SetupHelper #8: Test SetupHelper-beta

lennycb, 2022-01-01:
Just wanted to try a SetupHelper beta. Don't get it to work. Is there any trick? Have a test system with a RPI4 and VenusOS v2.80 ~ 33
[image]

kwindrem, 2022-01-01:
OOPS: PackageManager.py was missing from the blind install file (venus-data.tar.gz). Not sure how that happened but it is fixed now. v4.0~19 is on the beta branch.

lennycb, 2022-01-01:
I'll write again here, fits the topic better.
I'll try your adjustments in the RpiTemp.zip file right away.

Thank you very much for your extension. Saves me time because I no longer have to copy the setup script and try to rewrite it.

Now I just have to get to the helper (in the beta) to run. Unfortunately does not work. Where exactly does the .tgz have to be copied with the RPI4 so that it is installed automatically? It doesn't work on a stick. cannot install from there.

lennycb, 2022-01-01:
Gard tried, works for now.
I still have questions / comments:
1. When I load the directory from GIT (wget) the file permissions are not transferred correctly. Does that have to be adjusted in the script?
2. You packed the file "run" in the folder / service / RpiTemp. shouldn't that be on VenusOS> 2.80 in folder /opt/victronenergy/service/...? I once read something that all "own entries" in /service/... are deleted by a restart.

kwindrem, 2022-01-01:
The venus-data.tar.gz file goes on a USB stick or SD card. Don't unzip or unpack it. Put the card into the GX device and reboot twice, then remove the media.

Detailed instructions in the SetupHelper ReadMe.

I haven't had any issues with file permissions. My previous instructions used wget and everything worked since you are running scripts and services as root. What I recommend for a manual install is to go to a branch or one of the tags and download the tar.gz file and unpack it manually into /data on the GX device:

wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz |  tar -xzf -C /data

'latest' above could also be 'beta' if you want that version.

Where my setup scripts place the service directory depends on the Venus OS version. Prior to v2.80~10, it goes into /service. After that, it goes into /opt/victronenergy/service, BUT I also muck around in the /service directory and the overlay directory: /run/overlays/service to better accommodate installs and uninstalls updating the service without needing a reboot. I'm also careful about changing the files in the service directory if it's not needed as I found that caused issues where the service might not start/restart properly.

lennycb, 2022-01-02:
Moin, here are my test results:

Download SetupHelper beta with

`wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C / data`

then installed via `/data/SetupHelper/setup`

(Didn't work with the stick and the file venus-data.tar.gz. Just didn't want to install anything.)

The package manager was there and also ran.
Then I tried to install RpiTemp via the package manager.
Download -> OK
Installation ended with an error -> Setup cannot be executed
[image]

Checked the file rights via WinSCP, as expected, everything was only downloaded for reading.
[image]

Then I set the manual rights. and reinstalled via the package manager.
Installation got stuck.
[image]

Restart the RPI4 and reinstall it using the package manager
Installation -> OK
Temperatures are read and displayed.
[image]

My next attempt will be on a completely fresh installation.
Maybe a script should be found that checks all this 1wire stuff or, if necessary, installs it. But that definitely exceeds my abilities :-)

I am not yet sure how I give GIT the file permissions. I think I have read that this is probably always "reading" when transmitting from WIN computers. If you have a tip, please!

kwindrem, 2022-01-02:
I'm not sure how GitHub deals with file permissions from Windows. On my Mac, I set execute permission via the terminal then use GitHub Desktop to push to the repo. You may need to do a similar thing on Windows, possibly using the Properties, then Security, then Permissions dialog. But that's a guess since I no very little about Windows.

scp generally has a way to retain file permissions but don't know about WinSCP.

kwindrem, 2022-01-03:
SetupHelper v4.0~20 is now available on GitHub. It’s a major U/I change with some enhancements under the hood as well.

I’m hoping this will be the last beta.
—
Kevin

lennycb, 2022-01-03:
Hello, I've already tested it :-)
I've already noticed one thing. When I add a new package, the user and branch are forgotten. I had to enter that again under "active packages".

Apart from that, I got it right. As always, the mistake is 50cm in front of the monitor!
I had changed the rights and took an "old day". So the changes were not adopted by Git. Well, what the heck. I find git a bit exhausting.

During the days I tried to write a setup script for the additional modules before the actual RpiTemp installation.

The fact is, with a change from release to candidate, i.e. 2.7x to 2.8x. you bring Python3, PIP and some other bells and whistles. It's not that bad yet. One change back and EVERYTHING is gone. Not nice. How do you want to "automate" something like that? Have to think.

kwindrem, 2022-01-03:
Good catch on the add loosing user/branch.

If you select a known package it should bring the user/branch with them when you add. I’ll fix that.

SetupHelper should be reinstalling any packages that were installed before the downgrade. However, if the package is marked as not compatible with the lower version it can’t do the reinstall.

I’ll need to check a Venus firmware change/downgrade to see if something’s messed up in SetupHelper or PackageManager.

Thanks for testing. It really helps.

kwindrem, 2022-01-03:
v4.0~21 is up that fixes the GitHub info for a package add

I downgraded my RPI 4 to 2.73 from 2.80~33 and everything reinstalled. So we need to dig a bit deeper into your particular situation.

I suggest you send my the logs and I'll have a look:

/data/log/SetupHelper

data/log/PackageManager/current

lennycb, 2022-01-04:
Well, that's not "my problem". Personally, I might not care if it doesn't work with the 2.70. I don't use it anymore. But it would be cooler if it works under Python and Python3.

The fact is:
after a new installation an "opkg update" has to be done, otherwise you cannot install some modules.
If these modules are missing, PIP cannot be installed.
Without PIP there is no "pip install w1thermsensor".
And the DS18B20 does not work without "w1thermsensor".

Normally that's not a problem.
However, if you have to switch between 2.7x and 2.8x frequently (for whatever reason) you have to do this installation again and again. If you stay above 2.8x, everything is OK. In my opinion it has to do with the change to Python3.

I noticed that at the moment with the DS18B20. I hadn't connected any other sensors yet.
I hope you understand the meaning of what I am saying. Is made to the google translator.

virtuvas, 2022-01-05:
sorry for chipping in here as well Kevin,

I notice that lennycb edited and changed his github name on the beta package editor screen above. By default it comes with your name and I stupidly thought it was meant to be like that as it's on your github. 
Assuming I have to have my github name there, maybe it should be mentioned (or I'm simply thick and missed that!) or should be left empty with a note somewhere on the screen?
I'll wait for your feedback on the other issue (re OS Large menu disappearing) before I try it.

cheers

V.

lennycb, 2022-01-05:
@kwindrem 
I once dared to put something together.
You can look at this installation file
https://github.com/lennycb/RpiTemp/blob/main/setup_1wire

As I already wrote, a change from 2.7x to 2.8x or back is all that is needed.
What does the script.
- check whether the ReleaseType is correct and correct it if necessary
- opkg update
- install plugins (as described on the victron page)
- Get pip and install it (check which Python version beforehand)
- Install w1thermsensor
- add config.txt
- Copy overlayFile

After executing the file, the installation of RpiTemp with your PacketManager was possible without any problems (for release or candidate)

PS: I stole some of your code. :-)

kwindrem, 2022-01-06:
Great.

I avoid accessing the internet in the setup scripts I write because they would fail of there is no network connection. So maybe keeping this as a separate setup script is a good idea.

My code is there to use as others find it useful. Glad it helped.