kwindrem/SetupHelper #10: beta-testing v4.0~31

lennycb, 2022-01-07:
Hello Kevin, I had time to set up my test system again today.
I noticed the following, in bullet points, possibly without context:
- Cannot add "my" package. As a GIT version always only showed a "?"
- after manual installation of my package it was not displayed in the setup helper.
- the "Bachup Setup" button has no function
- Then when I wanted to add one of your packages, the system got stuck.
- after restart I had the SetupHelper version v4.0 ~ 17. for whatever reason

the two logs contain no errors.
/ data / log / SetupHelper
/ data / log / PackageManager / current

kwindrem, 2022-01-07:
Give me a link to your package and I'll try to add it to my system.

If you manually added by copying to /data, PackageManager should see it as an active package if it meets the criteria:
- needs a version file with a version of the form vX.Y (or vX.Y~Z for betas)
- does not contain certain strings in its name (mostly GitHub suffixes for tags and versions (i.e. -latest)

For PackageManager to install it, an executable file named setup must also exist. It needs to return specific exit codes to control behavior after install. See the comments at the beginning of PackageManager.py for details.

Not sure how you got v4.0~17 unless you have that version on a USB stick inserted in the GX device. Maybe a an old venus-data.tar.gz?

In order to use Backup & Restore settings you need a USB stick or SD card inserted. Then Backup settings should appear in the menu. After you run the backup, the Restore settings button should appear.

When you day "system got stuck" what did it do and what was displayed in the Package editor and/or on the Package Manager menu status line?

lennycb, 2022-01-07:
actually, on the stick that I plugged in for the backup, there was still a version of venus-data.tar.gz on it. That could explain the matter with the version and the crash. Will have been working in the background and I tried to install a package.
Just strange, if I wanted to install "intentionally" over the file, it did not work. Does not matter.

I have now updated to v4.0~31 via the PacketManager, unfortunately the PacketManager does not start anymore after completion.

now to my package.
I deleted it manually because the PackageManager.py didn't want to start. Obviously he didn't like the version number (v1.0b1)

manual start via the console:
```
root@raspberrypi4:/data/SetupHelper# ./PackageManager.py
WARNING:>>>> PackageManager starting
Traceback (most recent call last):
  File "./PackageManager.py", line 3142, in <module>
    main()
  File "./PackageManager.py", line 3076, in main
    package.UpdateFileFlagsAndVersions ()
  File "./PackageManager.py", line 1769, in UpdateFileFlagsAndVersions
    self.SetInstalledVersion (installedVersion)
  File "./PackageManager.py", line 1324, in SetInstalledVersion
    self.InstalledVersionNumber = VersionToNumber (version)
  File "./PackageManager.py", line 485, in VersionToNumber
    versionNumber += int ( parts[2] ) * 1000
ValueError: invalid literal for int() with base 10: '0b1'
root@raspberrypi4:/data/SetupHelper#
```

but even after deleting RpiTemp the service would not start.

-> So back to the beginning :-)
- new image
- new SetupHelper installation (-> via console)
- GiuMods installed (-> via PackageManager)
- RpiGpioSetup installed (-> via PackageManager)
  here I had to wait 7.9 minutes (download starts in x.x minutes)
- RpiTemp installed (-> via PackageManager) OK

So all together it was probably due to the version number and probably also that I bent the system correctly with the cleanup and troubleshooting.

my link to the repo. I'm still fine-tuning there :-)
https://github.com/lennycb/RpiTemp.git

PS. tell me, do you know how to build an environment to compile? I did not find it in the VenusOS pages. They don't answer in the forum either.

I wanted to try something here.
https://github.com/victronenergy/dbus-ble-sensors.git

kwindrem, 2022-01-07:
I was able to add your package to my PackageManager and manually download and install it. It's running!

I'll make some changes to the version checking to be more flexible: supporting strings like v1.0b1 and v1.0.1. Should be part of the next release. I've never really liked the v1.0~3 format but that's what Victron uses for some reason. GitHub won't except it as a valid tag so may change to a more common format for my stuff.

kwindrem, 2022-01-08:
Regarding PackageManager not running: send me it's log: /data/log/PackageManager/current or at least the last 20 or so lines.

lennycb, 2022-01-08:
> Regarding PackageManager not running: send me it's log: /data/log/PackageManager/current or at least the last 20 or so lines.

unfortunately that is gone after the reinstallation. Will think of the logs when "problems" arise again. Everything is currently running!

kwindrem, 2022-01-09:
I mocified the version number processing to accept a wider range of version formats including v1.2b3. I don’t think PackageManager will crash any more even if I can’t figure out the version.
 v4.0~32

Jsrivet, 2022-01-15:
I don't know the reason when running one of version 4 my screen goes white. I reverted to 3.15 and no longer have a problem.

I have a log in PackageManager which gives me this message:

@4000000061e262cf35cf1fa4 WARNING:>>>> PackageManager v4.0 starting
@4000000061e26d4b0ebfa85c WARNING:>>>> PackageManager v4.0 starting
@4000000061e26da80d1649d4 *** CCGX booted (0) ***
@4000000061e26db02fe6f65c WARNING:>>>> PackageManager v4.0 starting
@4000000061e270ee0d0a5354 *** CCGX booted (0) ***
@4000000061e272f51b23e61c WARNING:>>>> PackageManager starting

kwindrem, 2022-01-15:
The white screen indicates an issue with the GUI so the appropriate log would be /data/log/SetupHelper (no /current on this one).

Thanks for reporting. I can't reproduce here so will need the log from your system.

Jsrivet, 2022-01-15:
The problem comes from V2.80-33 large-24 every time I switch to this version the screen goes white but the other functions are correct. I uninstall SetupHelper via SSH and restore V2.80-36 and SetupHelper and GuiMods come back normal. I am attaching the SetupHelper log there are several installs and uninstalls. Another question is it possible to restore a backup version via SSH or console? Thank you

@4000000061e2fa3300e39f7c SetupHelper/setup: installing PackageManager service
@4000000061e2fa330536e46c SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@4000000061e2faa203732bcc SetupHelper/setup: removing PackageManager service
@4000000061e2faa20926ef04 SetupHelper/setup: removing SetupHelper/setup from reinstallScriptsList
@4000000061e2facb206b33b4 SetupHelper/setup: completed
@4000000061e2fad31bf887b4 SetupHelper/setup: completed
@4000000061e2fbee23be9ca4 SetupHelper/setup: installing PackageManager service
@4000000061e2fbee284011cc SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@4000000061e300e933dd84ec SetupHelper/setup: removing PackageManager service
@4000000061e300e93945d7a4 SetupHelper/setup: removing SetupHelper/setup from reinstallScriptsList
@4000000061e300fb25f9179c SetupHelper/setup: completed
@4000000061e301021786ed24 SetupHelper/setup: completed
@4000000061e301160b5d04fc SetupHelper/setup: installing PackageManager service
@4000000061e3011611b94504 SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@4000000061e301483434ccac SetupHelper/setup: removing PackageManager service
@4000000061e3014839c4f2b4 SetupHelper/setup: removing SetupHelper/setup from reinstallScriptsList
@4000000061e301b333eb679c SetupHelper/reinstallMods: reinstallMods starting
@4000000061e301b505ac7614 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e301b93a45619c SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e301c12525068c SetupHelper/reinstallMods: reinstallMods complete
@4000000061e3022621ef8dac SetupHelper/setup: creating file set for v2.80~36
@4000000061e3022d2cc6650c SetupHelper/setup: installing PackageManager service
@4000000061e3022d31e3629c SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@4000000061e303032f916fd4 GuiMods/setup: installing GuiMods files
@4000000061e3030424bb1574 GuiMods/setup: ++ GuiMods installed
@4000000061e30304269167cc GuiMods/setup: adding GuiMods/setup to reinstallScriptsList
@4000000061e3030428604fb4 GuiMods/setup: completed
@4000000061e307e2327b262c SetupHelper/reinstallMods: reinstallMods starting
@4000000061e307e4046f567c SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e307e93704770c SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e307f41f5d888c SetupHelper/reinstallMods: reinstallMods complete
@4000000061e30877314b3d64 SetupHelper/reinstallMods: reinstallMods starting
@4000000061e3087836b12304 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e3087e22b88acc SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e308881820db64 SetupHelper/setup: installing PackageManager service
@4000000061e3088e198cdafc GuiMods/setup: installing GuiMods files
@4000000061e308943508705c GuiMods/setup: ++ GuiMods installed
@4000000061e3089509e50ee4 SetupHelper/reinstallMods: reinstallMods complete
@4000000061e3096b1072328c SetupHelper/setup: removing PackageManager service
@4000000061e3096b191d4494 SetupHelper/setup: removing SetupHelper/setup from reinstallScriptsList
@4000000061e309a336f5ec64 SetupHelper/reinstallMods: reinstallMods starting
@4000000061e309a43a9c2874 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e309a82bc58b24 SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e309b01adf91ec SetupHelper/reinstallMods: waiting for dBus settings
@4000000061e309bb044e8424 SetupHelper/reinstallMods: reinstallMods complete

kwindrem, 2022-01-15:
If you are talking about a package backup, there aren't really any stored on the system. However, you can use PackageManager to change the version that gets downloaded the do a manual or automatic download and install.
The normal "tag" for my packages is "latest" or ("beta" if you are beta testing). There are also tags for each version so you could for example set the branch/version to v3.15.

Any the problems you are experiencing should be fixed now with v4.1

Jsrivet, 2022-01-15:
Thank you problem solved with V4.1. very good work really appreciate.

closed 2022-01-19 by kwindrem