kwindrem/SetupHelper #13: Venus OS 2.84: Install Manger not showing

seanslife, 2022-02-19:
Hi there, I've tried to use blind install on SetupHelper but after two reboots there is nothing added to the standard Venus menu. 
- I copy the venus-data.tar file to a freshly formatted USB stick
- I reboot twice waiting for GUI to fully load
- - When I scroll to the bottom of the Menu there is no extra line added (it ends I with I/O)
- -I did notice an extra file created on the USB stick first time I tried it, but not since
- I've reformatted USB stick several times, used a different USB stick but no joy. 
- The version of Venus OS I'm using is the latest (v2.84) running on a Rpi3B+
Do you have any tips for me to try? Cheers Sean

kwindrem, 2022-02-19:
The file needs to be venus-data.tgz to be unpacked. venus-data.tar will not be recognized by the script that Victron provides for such blind updates.

Try not unzipping the file. Copy it straight from the repo to the USB stick. Note on Mac OS, you may need to disable Open "safe" files after downloading in Safari Preferences. There might be something similar in Windows too.

seanslife, 2022-02-19:
Yep, that solved it. I just copied directly to the USB stick and the .tgz file appeared there just fine. FYI, on second reboot the RPi hung on reboot, so after waiting 15mins I disconnected and it powered up no problems, getting through to the GUI where I could see and use the Package Manager. Have now installed GUImods and RpiScreen. Great work on these augmentations - you should be working for Victron!

Twabraun, 2022-03-13:
I have the same problem; V2.84; Windows format new FAT32; I tried it several times; Reboot 2 time before I took the stick out. 

What can I do?

Twabraun, 2022-03-13:
I made it, Super perfect thanks. This is what I needed.

GlisGlis47, 2022-03-23:
How did you made it? Got the same problem.

morphias2004, 2022-03-28:
> I made it, Super perfect thanks. This is what I needed.

Having the same issue.

What did you do to get it working?

morphias2004, 2022-04-11:
@kwindrem are you able to assist please?

Do we need to roll back from 2.84 to 2.80 for it to work?

I'm only having the issue on a Cerbo GX. It went fine on an RPi 3B+.

ducloslaurent, 2022-04-19:
Hello kwindrem,
Thanks for the work done for the Venus OS customization.
I try to follow the SetupHelper installation guide but even if I try several USB key, different partition type (FAT/FAT32/NTFS) and even if it is the .tgz (not the .tar) I put on the USB root file, impossible to see th SetupHelper menu installed on the 2.84 Venus OS.
-> tail -50 /data/log/SetupHelper | tai64nlocal command result via root access using Putty is 
tail: can't open '/data/log/SetupHelper': No such file or directory
tail: no files

-> any advice, recommendation?
[image]

kwindrem, 2022-04-19:
I'm not sure what is going on on your system. Let me explain the process so you can see where things might be failing.

The blind install process is as follows:

On the first reboot: unpacks the .tgz file to /data/SetupHelper-blind (folder) and /data/rcS.local (file).

If that part of the blind install succeeds, you should see those two components in ls -al /data

The second reboot executes the rcS.local file which runs /data/SetupHelper-blind/blindInstall

The blindInstall script then moves SetupHelper-blind to SetupHelper then runs the SetupHelper/setup script which installs PackageManager.

If you aren't seeing the log file, then it is likely the setup script is never being executed.

It does take two reboots to get everything done so if you haven't rebooted a second time, I'd try that first.

You can also install manually from the command line:
	wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C /data
	mv /data/SetupHelper-latest /data/SetupHelper
	/data/SetupHelper/setup

ducloslaurent, 2022-04-19:
Blind install process works with USB key and VEnus OS v2.66 -> Package Manager installed succesfully.
Try with fresh Venus Os install with v2.85 version and the same USB key, reboot almost 4 times -> no Package Manager installed
Putty access with root login, using the provided command line -> installation done!
Many thanks for the support and quick answer

kwindrem, 2022-04-19:
Just tried the bind install on my PI test system and it worked fine. Venus OS v2.85. 

What platform are you running Venus OS on?

ducloslaurent, 2022-04-19:
Using Multiplus 2 48/3000 (not GX device) with 3600Wc (12 x 300Wc panels) solar panel plugged on 3 micro inverter plugged on the AC out 1 of the Multiplus 2 and 14kw LifePo4 (16 cells each 280Ah) monitored with a JK BMS

morphias2004, 2022-04-20:
I has no issue at all doing this on an RPi 3B+ with v2.84 installed.

Where I'm running into trouble is on a Victron Cerbo GX unit that was on v2.84.

I just tried in on a unit that was still on v2.73 and it worked perfectly.

I have tried downgrading the problem unit to v2.80 and trying the blind install again without success and then downgrading to v2.73 with the same failed result.

I've just enabled SSH and connected to it.

The blind install has got as far as unpacking the compress tgz file by the looks of it as the SetHelper directory is present:

```
root@einstein:/data# ls -al /data
drwxr-xr-x   15 root     root          4096 Apr 20 03:08 .
drwxr-xr-x   20 root     root          1024 Apr 20 03:08 ..
-rwxr-xr-x    1 501      dialout        276 Jan 11 23:32 ._rcS.local
drwxr-xr-x    5 501      dialout       4096 Mar 28 03:12 SetupHelper
drwxr-xr-x    3 root     root          4096 Apr 20 03:22 conf
drwxr-xr-x    2 www-data root          4096 Apr 20 03:07 db
drwxr-xr-x    2 root     root          4096 Jan  1  2010 etc
drwxr-xr-x    4 root     root          4096 Jan  1  1970 home
drwx------    2 root     root          4096 Apr 20 02:47 keys
drwxr-xr-x   37 root     root          4096 Apr 20 03:07 log
drwx------    2 root     root         16384 Jan  1  1970 lost+found
drwxr-xr-x    3 root     root          4096 Mar 28 03:12 setupOptions
drwxr-xr-x    3 root     root          4096 Jan  1  1970 themes
drwxr-xr-x    3 root     root          4096 Jan  1  1970 var
drwxr-xr-x    2 root     root          4096 Apr 20 03:07 venus
drwxr-xr-x    2 root     root          4096 Apr 20 02:47 vrmfilescache
root@einstein:/data#
```

 and ran `tail -50 /data/log/SetupHelper | tai64nlocal`

and got the following:

```

root@einstein:~# tail -50 /data/log/SetupHelper | tai64nlocal
2022-03-28 03:50:43.286465500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-28 03:50:50.349221500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-03-28 03:57:32.407543500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-03-28 03:57:33.363039500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-28 03:57:36.542156500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-28 03:57:42.925615500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-28 03:57:47.395519500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-28 03:57:54.663090500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-03-29 02:10:39.338293500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-03-29 02:10:40.568921500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:10:43.677133500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:10:50.921023500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:10:57.786474500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-03-29 02:26:55.350957500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-03-29 02:26:56.405936500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:26:58.434610500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:27:05.771882500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:27:09.992957500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-29 02:27:16.151924500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-03-31 07:48:31.346355500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-03-31 07:48:32.369789500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-31 07:48:34.707490500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-31 07:48:41.339518500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-31 07:48:46.488427500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-03-31 07:48:54.057200500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-04-20 02:45:39.367042500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-04-20 02:45:40.438177500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 02:45:43.922141500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 02:45:50.445642500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 02:45:54.536516500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 02:46:01.635012500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-04-20 02:59:59.911270500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
2022-04-20 03:00:01.187310500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:00:03.773857500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:00:11.818422500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:00:19.548540500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
2022-04-20 03:07:50.638824500 SetupHelper/reinstallMods: /data/reinstallScriptsList file not found
[... cut ...]
2022-04-20 03:08:10.503807500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:08:13.479182500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:08:18.431892500 SetupHelper-blind/blindInstall: waiting for dBus settings
2022-04-20 03:08:23.716642500 SetupHelper-blind/blindInstall: blind install did NOT update SetupHelper (probably an older version)
```

So I'm thinking the downgrading had no effect because the SetupHelper directory was already present.

Manually running `/data/SetupHelper/setup` has resolved it.

kwindrem, 2022-04-20:
The messages "blind install did NOT update SetupHelper" indicates the blind version is not newer than the currently installed version. I did this so that if someone leaves the USB stick plugged in it won't attempt to install again and again. There may be a case where SetupHelper directory is present but SetupHelper is not actually installed.

I'll give this some thought and come up with a better way to insure the blind install insures SetupHelper is actually installed.

Thanks for the log. It really helps.