kwindrem/SetupHelper #16: After the latest change, my 2 RPI running 2.85 are stuck on a constant GUI reboot.  I SSH and uninstalled setuphelper and the issue went away

pedrorampolla, 2022-05-03:


pedrorampolla, 2022-05-03:
New load of 2.85 and either on blind install or via script, the problem starts as soon as setuphelper is installed and goes away as soon as it is removed.

RobDuthie, 2022-05-03:
Yes it crashes gui mods have to reload new image to start again can't access as it loops and restarts.

kwindrem, 2022-05-03:
Working fine here. Can someone send me gui, SetupHelper and PackageManger log files.

kwindrem, 2022-05-04:
Should be fixed now in v5.23. If you have auto downloads and installs enabled, the problem will correct itself.

RobDuthie, 2022-05-04:
Permission denied? whne reinstall setup helper?

RobDuthie, 2022-05-04:
Still doing it missing ver file sets 2.85

closed 2022-05-04 by pedrorampolla

kwindrem, 2022-10-11:
Rob, are you still having issues with GuiMods v5.23? I deployed that last night. Works here on v2.85 firmware.
—
Kevin