kwindrem/SetupHelper #17: Can't install on 2.90~10

SergioRius, 2022-06-04:
I had previously updated to 2.90~10 (large) thus loosing all the mods.
Today I tried to reinstall the SetupHelper using the manual procedure but there's some problem with PackageManager and the install seems not functional.
```
root@einstein:/data# wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C /data
root@einstein:/data# mv /data/SetupHelper-latest /data/SetupHelper
root@einstein:/data# /data/SetupHelper/setup
creating root setup options directory /data/setupOptions
creating package options directory /data/setupOptions/SetupHelper
creating file set for v2.90~10

This package provides support functions and utilities for Venus modification packages
Packages are automatically reinstalled following a Venus OS update
All actions are controlled via /Settings/PackageMonitor in the GUI
  Packages may be automatically updated from GitHub or a USB stick
  Previously uninstalled packages can also be downloaded an installed

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i
installing PackageManager service
svc: warning: unable to chdir to /service/PackageManager: file does not exist
adding SetupHelper/setup to reinstallScriptsList
adding SetupHelper reinstall script to /data/rcS.local
Restart the GUI now (y) or issue a do it manually later (n): y
restarting GUI ...
root@einstein:/data#
```

```
root@einstein:/data/SetupHelper# ./setup

This package provides support functions and utilities for Venus modification packages
Packages are automatically reinstalled following a Venus OS update
All actions are controlled via /Settings/PackageMonitor in the GUI
  Packages may be automatically updated from GitHub or a USB stick
  Previously uninstalled packages can also be downloaded an installed

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i
restarting PackageManager service
svc: warning: unable to chdir to /service/PackageManager: file does not exist
svc: warning: unable to chdir to /service/PackageManager/log: file does not exist
svc: warning: unable to chdir to /service/PackageManager: file does not exist
svc: warning: unable to chdir to /service/PackageManager/log: file does not exist
completed
root@einstein:/data/SetupHelper# ll service
drwxrwxr-x    3 root     root        4.0K May  4 23:17 .
drwxrwxr-x    5 root     root        4.0K May  4 23:17 ..
drwxrwxr-x    2 root     root        4.0K May  4 23:17 log
-rw-rw-r--    1 root     root        6.0K May  4 23:17 .DS_Store
-rwxrwxr-x    1 root     root          62 May  4 23:17 run
root@einstein:/data/SetupHelper#
```

kwindrem, 2022-06-05:
I can not explain what is going on. 

/service/PackageManager should be created during SetupHelper install and the "installing PackageManager service" is output just before that happens. The second setup shows "reinstalling PackageManager service" indicating that the first install at least partially created it. but the chdir errors indicate that /service/PackageManager probably doesn't exist. You could check the /service directory to see if it does.

In Venus OS v2.90, /run/overlays/service is mounted on top of /service. This is done because the root file system is mounted read-only. (SetupHelper remounts rootfs read-write for other reasons so this is redundant.)

jsalbre, 2022-06-05:
I got the exact same errors, but upon rebooting my Cerbo twice everything appeared to work just fine.

kwindrem, 2022-06-05:
I tried a manual instal on v2.90~10 large (although the version info doesn't indicate large) and SetupHelper installed and PackageManager started.

I did get an error "svc: warning: unable to control /service/PackageManager: file does not exist" but everything was working.

I'll look into the error.

To be clear, I did not get the chdir warning and PackageManager was running.

kwindrem, 2022-06-05:
The errors appear to be due to a timing issue. The next release of SetupHelper will include delays while installing a service to prevent the errors.