kwindrem/SetupHelper #18: Can’t get around the gui white screen on 2.73

pedrorampolla, 2022-06-15:
I’m using firmware 2.73 since it’s the latest one to support multiple UART BMS.   Every time I install setuphelper and the GUI reboots, I get the white screen.  If I uninstall, the GUI goes back to normal.  Followed the instructions, (blind install, SSH, reflashing the firmware (2.73) and starting over.

pedrorampolla, 2022-06-15:
It was working on the latest firmware, but I had to revert in order to enable the BMS functionality.

kwindrem, 2022-06-15:
Should be fixed now. If you have auto downloads and updates enabled, the new version should update automatically.

You can also do a blind instal with a fresh download of venus-data.tgz

The issue was an unsupported mechanism to unmount USB/SD media that was apparently added in v2.80. Used the old way for v2.73 and before.

pedrorampolla, 2022-06-15:
That fixed it.  Thanks so much.

closed 2022-06-15 by pedrorampolla