kwindrem/SetupHelper #19: 123SmartBMS-Venus appearing on Active Packages whitout verion N.

GianFabris, 2022-07-13:
Hello, 
My set-up is:

- Multiplus II GX 48V 5000VA

- LiFePo4 200 Ah 48V battery pack

- 123 smart BMS

1. I have updated the Multiplus firmware to the latest version 497
2. I have updated the firmware of the GX unit to v2.87
3. I downloaded venus-data.tgc and copied to USB memory memory stick
4. Boot the system with memory stick
5. Reboot the system from GUI
6. Removed the memory stick while rebooting

What is appearing on Package manager under "Active packages" is a line **123SmartBMS-Venus** without indication of any stored or installed version.
Is that correct?

From this point how should I detect if 123 BMS is working?

GianFabris, 2022-07-13:
[image]

AlbertronicDev, 2022-07-13:
Hi,

Where did you download the venus-data.tar.gz? The latest version can be found here: https://123electric.eu/products/123smartbms-to-usb/

Please put it on a USB, reboot twice and check again. If none works, please let us know.

GianFabris, 2022-07-14:
Hi,
Thank you for your reply.
I downloaded the venus-data.tar.gz from here https://github.com/kwindrem/SetupHelper
I will try download the file from your link and let you know.
Thank you for support.

Gianluca Fabris

kwindrem, 2022-07-14:
https://github.com/kwindrem/SetupHelper does not include the 123SmartBMS package. So there was nothing to install besides SetupHelper itself.

The one line on the 123 site DOES have the BMS package in addition to SetupHelper so that's the one you want.

With SetupHelper installed and your GX device connected to the internet, you could also use PackageManager (menus at the bottom of the Settings menu) to download the 123SmartBMS package. But I think the blind install provided by 123electric (not the one in SetupHelper) is fewer steps.

GianFabris, 2022-07-14:
[image]

[image]

Hi,
Installing the venus-data.tar.gz provided by 123electric , i can see now the BMS parameters.
Can you coinfirm that the version 1.3 provide also control not only displayng?
Thank you foir your support

Gianluca

AlbertronicDev, 2022-07-15:
Hi Gianluca,

v1.3 also can control the Victron system. Do not forget to enable DVCC and configure the Multiplus to listen to this. See "How to control a Victron with GX" at https://123electric.eu/products/123smartbms-gen3/

GianFabris, 2022-07-15:
Thank you for the support

closed 2022-07-15 by GianFabris