kwindrem/SetupHelper #23: Feature Request

tlewis99, 2022-08-31:
Is is possible to include in the backup and restore settings:
I/O / Analog tanks settings
then any customizations about Analog tanks made from the Device menu?

Let me know your thoughts.

Thanks,

Tim

kwindrem, 2022-08-31:
I think this might be possible but each of the GX device platforms will have different parameters. By inspection, Cerbo has .../adc_builtin0/1/... .../adc_builtin0/4/... for tanks. I'm not sure what other platforms would have. I would need a complete list of dbus Settings parameters to include them all. Analog hats on the Raspberry PI would probably also have dbus parameters different from other platforms and may in fact change based on the analog interface.

Then there are separate setting parameters for the GX Tank.  I would not be able to include these in the backup and restore because the dbus path includes the GX Tank's serial number which is not consistent between systems.

Unfortunately other external tank sensors utilize the external device's name (or other field) instead of having something in dbus Settings. So these would also not be included.

So if I did this, it would support ONLY Cerbo built-in tanks.

tlewis99, 2022-08-31:
I understand about the different platforms.  In my case, I consistently deploy on a CerboGX  with analog tank 1 which includes a custom name and Ohm settings for trailers that all use a 30gal tank for a diesel generator.  I’m hoping not to have to manually input the tank info on every unit I deploy.

kwindrem, 2022-08-31:
I have added Cerbo tanks (and temps) -- including alarms to backup and restore. It's currently on the beta branch so change the SetupHelper branch in PackageManager to get that version for testing. 

You can also download from GitHub and install manually if you prefer. Make sure you select the beta branch before downloading!

I'll release as soon as you report back and I've fixed any bugs.

tlewis99, 2022-08-31:
Fantastic,  I’ll test tomorrow.

tlewis99, 2022-09-01:
I have tested with the following:
V2.90~26 Einstein firmware
V4.23~1 SetupHelper

When you first go into Settings/ I/O /Analog inputs Tanks Level input 1 is off
[A screenshot of a computer  Description automatically generated with medium confidence]

Randomly I toggled tank 4 and tank 1 also came on making me thing the GUI just didn’t refresh properly

[A screenshot of a computer  Description automatically generated with medium confidence]

Looking at the devices page it looks like the customizations are restored
[Graphical user interface, application  Description automatically generated]

When I drill down into Geni Diesel 30gal setup, I realize not all setting are restored.  It sets the Standard to custom but for the values it uses the European default values of 0 and 180.  You can see from the attached settingBackup this should be 250 and 35
[A screenshot of a computer  Description automatically generated with medium confidence]

FYI, please let me know if you need a CerboGX for testing.  You can email me outside of github with shipping info and I would send you one you can keep.

Tim

tlewis99, 2022-09-01:
Here is a link to the settingBackup file:
https://collectiveinnovations0-my.sharepoint.com/:u:/g/personal/tlewis_collectiveinnovations_com1/EQASD48o_yJEgG7sAKPDkuMBz0jepkIAHdKCYjOZvupeUQ?e=hacAL5

Here is a link to a PDF version of the above email with the embedded screenshots:
https://collectiveinnovations0-my.sharepoint.com/:b:/g/personal/tlewis_collectiveinnovations_com1/EYnu-owSxUtOsCUcjNtLzBkBjqBt8AxgoT8zN8R-17vFSg?e=JAvhJj

kwindrem, 2022-09-01:
I did some tests with dbus-spy and unless the standard is set to custom, changes to the sensor empty/full values are rejected. In fact, setting to US or European sets the empty/full values to defaults for those standards !!!

I looked at my settingsList and the standard is set AFTER the values. I reordered things and ran a test and the custom empty/full values seem to be restored properly.

v4.23~2 is up on GitHub now. Give it a shot.

tlewis99, 2022-09-01:
The custom Ohm settings work now without a reboot.  Prior to reboot, if you look in settings\i\o\analog tanks, tank 1 is still off.  It must be on because the tank shows up on the Device screen.  After a reboot, tank 1 does show on under analog tanks, all without any user intervention.

kwindrem, 2022-09-01:
I tried this on my Cerbo and the tank shows in the GuiMods enhanced overviews without a reboot. And as you say, it is showing up in the device list. I also discovered what you did -- that the tank input switch shows off in the Analog inputs menu. I changed the Function dbus Setting for that input and the switch setting did not change. I also found I could no longer change the switch in the menu.

I restarted the GUI (kill all gui) and the tank then shows as on.

It appears that menu is not refreshing when parameters change from something outside that menu. I don't fully what is going on in that menu or how to fix it.

tlewis99, 2022-09-01:
My two cents, I think it is a very minor nuisance and if I wasn’t actually testing your release I wouldn’t have noticed.  I’m also not bothered by having to do a reboot after restoring settings.

kwindrem, 2022-09-01:
Agreed on the tank on/off issue. It turns out if you change ANY switch setting in this menu, it refreshes the other switches also.

If you have more testing to do, I'll hold off releasing, otherwise, I'm ready on my end.

tlewis99, 2022-09-01:
I’m good,

Thanks!

tlewis99, 2022-09-02:
Kevin,

No rush,  just making sure you’re not waiting on me?

Tim

kwindrem, 2022-10-11:
No. Just got sidetracked on other things. I’ll release now.
—
Kevin

mjelkins, 2023-06-18:
An idea for the Mopeka devices, so in my Caravan I have a Mopeka LPG device. It is working fine. I've also got a back order for the Water Tank Monitors (for my fresh water). It would be great to calculate (guesstimate) the lifespan of these tanks as to when they will get to empty... rather than get a cold shower, or simply no water. Resolution to an hour based on the last day or two would be perfect.