kwindrem/SetupHelper #24: SetupHelper generates errors in message

Aurgelme, 2022-09-16:
SetupHelper generates error every minute in /data/log/messages

`raspberrypi2 daemon.err dnsmasq[702]: failed to send packet: Address family not supported by protocol`

Aurgelme, 2022-09-16:
System pi3+ and venus-os 2.90
Error messages go away when I uninstall SetupHelper, then installed GuiMods and ShutDownMonitor manually and still no error messages.
Reinstalled SetupHelper and the error messages reappear every 60 seconds

kwindrem, 2022-09-16:
The logs don't show anything out of the ordinary so not sure where to go from here. I'm not seeing this on my RPI 4.

Uninstalling SetupHelper will prevent PackageManager from running so that's probably where the issue is. Try reinstalling SetupHelper and verify the errors are occurring.

Then shutdown PackageManager:

svc -d /service/PackageManager

Then check to see if the errors are occurring again.

Maybe you could send more of the messages log so I can look to see if there are any other clues.

Aurgelme, 2022-09-16:
I have gone through the logs and only find that it is the SetupHelper that triggers (the network error?)

When I set up ethernet in the Remote Console to manual, without changing any of the values, the error disappears. (dns 192.168.1.1)
If I manually set a new dns server (8.8.8.8) still no errors.

Select automatic network configuration again, reboot and it still says 8.8.8.8 in the network configuration in RemoteConsole after reboot and the error message returns

Selecting manual network configuration again, still without changing any values, and the error is gone

I don't know if this is an error that occurs because of my network (LTE 4G), so the problem looks solved from my side.

Your effort is appreciated

closed 2022-09-19 by Aurgelme