kwindrem/SetupHelper #28: addition function package: 1.SDM630,  2.JK-BMS , adding turn Display view 180° by a GUi-Switch.

alexseuf, 2022-10-09:
Hello,

1. is it posible to incloude Eastron SDM630/SDM230 Modbus Energy Meter support as a package into setupHelper?
https://community.victronenergy.com/questions/85564/eastron-sdm630-modbus-energy-meter-community-editi.html

2. more imprtent than 1.: because JK-BMS become 1th choise because of active balancing abliity:
  Add https://github.com/Louisvdw/dbus-serialbattery as a function package?

3. turn Display view 180° by a GUi-Switch. This is last manual step which has do bee done with putty: see [25:42](https://www.youtube.com/watch?v=Fi6vhC5_YPA&t=1542s) - Rotate LCD Raspberry Pi Config

Thank you for your perfect work

kwindrem, 2022-10-09:
None of these items is within the scope of SetupHelper.

Energy meter support would need a separate package written which could then be managed by PackageManager.

Flipping the display is easy and persists a reboot so really only needs to be done once. But I can look at an option to RpiDisplaySetup's setup script (manual run).

closed 2022-10-09 by kwindrem

alexseuf, 2022-10-09:
Thank you for your answere.
my english is not really perfect, do I understand you right, that you do
not accept point one and two..
and I should write topic three into the feature requests of rpiDisplaySetup?

Thank you

Alexander Seufert
Germany

kwindrem, 2022-10-09:
Added screen rotate to RpiGpioSetup (v2.10)