kwindrem/SetupHelper #29: Adding 3.5 Screen to Pi

Kevlar1234, 2022-11-01:
Hello Kevin

I am her from following Andy's Off Grid Garage.

I am trying to add a 3.5 inch display to my pi an cannot get the drivers to install.

Can you add it to your "Download Helper" as an option to install? 
or can you provide any guidance on how to install the drivers with the Venus OS?

Instructions that came with the 3.5 in display:

Product description 
Effekt video: https://youtu.be/CIizY3jXuAA 

LCD Type: TFT; LCD interface:SPI (supports up to 125MHz SPI input); Touch Screen Type:Resistive; Touch Screen Controller:XPT2046; 
Resolution: 320*480 pixel;
Protective Case with heat sinks make sure Raspberry pi work efficient and safe, full access to all controls and features. 

Driver installation:
Step 1: install the Raspbian official mirro: 
1. Download the latest Raspbian mirror from official 
2. Use SDFormatter to format Micro SD card 
3.Use Win32DiskImager to burn the official image to Micro SD Card 

Step 2: Connect the screen with your Raspberry Pi 

Step 3: Install Driver for Raspberry Pi 
1.Prepare keyboard and mouse with your Pi , connect internet. 
2.Execute the following commands 
sudo rm -rf LCD-show
git clone 	 
chmod -R 755 LCD-show 
cd LCD-show/ 
Sudo ./MHS35-show 

If you have difficulty installing the driver, or if you still can't use the display properly after installing the driver, please try our pre-install driver image for tested. 

For Raspberry Pi 3 B/B+, Pi Zero/W
RetroPie game system: https://drive.google.com/open?id=1KQpI3sJ6l3NNQTEruPT83050_Sl4IDsi 

For Raspberry Pi 3 B/B+
Raspbian: https://drive.google.com/open?id=1rQQyoNXkxVH8vcLW_tUKWsMs4fgfJacj 
Kali: https://drive.google.com/open?id=1NoHoxBSIP4i8WKCJo761K_-DcYh1IZ7k 

For Raspberry Pi 3 & Pi 2 (Notice: Ubuntu system can’t support Raspberry Pi 3 B+ board)
Ubuntu: https://drive.google.com/open?id=1zZRt4SCdwn1U0SYDCMxLNcFud0olY7Pi 

Rotate the display Tutorial: 
https://drive.google.com/open?id=1Xh-RW62XNluIBA8vyOAJroiUjY8XvOk

Thank You 
Kevin from Canada

kwindrem, 2022-11-01:
This really would be part of RpiDisplaySetup but I'd rather not make additions for a specific display. I'd be at this constantly for each new display. Plus I'd need to buy each display in order to debug the code.

The drivers are not setup well for download and automatic install. You should be able to download them to the RPI running Venus OS with the command

wget https://github.com/goodtft/LCD-show

then following the instructions you referenced above.

closed 2022-11-01 by kwindrem

Kevlar1234, 2022-11-01:
Hello Kevin

Thank you that line worked.

When I get to the below line I get the below error.

root@raspberrypi2:~# cd LCD-show/
-sh: cd: LCD-show/: Not a directory
root@raspberrypi2:~#

is there a different cd command I should be using?

tks
Kevin CDN

kwindrem, 2022-11-01:
Where you run these commands makes a difference. You need unix command line experience to figure this out. It's not something I can really help you with via this forum.