kwindrem/SetupHelper #30: ERROR: v3.00~7 PageSettings.qml no replacement file

CzarofAK, 2022-11-17:
root@CerboGX:~# /data/SetupHelper/setup
creating file set for v3.00~7
ERROR v3.00~7 PageSettings.qml no replacement file
ERROR: incomplete file set for v3.00~7 - can't continue
root@CerboGX:~# 

Guess the files for the new release are missing

CzarofAK, 2022-11-17:
Tried also v2.92

root@CerboGX:~# /data/SetupHelper/setup
creating file set for v2.92

This package provides support functions and utilities for Venus modification packages
Packages are automatically reinstalled following a Venus OS update
All actions are controlled via /Settings/PackageMonitor in the GUI
  Packages may be automatically updated from GitHub or a USB stick
  Previously uninstalled packages can also be downloaded an installed

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i
ERROR: no replacement file for PageSettings.qml
installing PackageManager service
svc: warning: unable to chdir to /service/PackageManager: file does not exist
removing PackageManager service
svc: warning: unable to chdir to /service/PackageManager: file does not exist
svc: warning: unable to chdir to /service/PackageManager/log: file does not exist
installation failed - package uninstalled - exiting
root@CerboGX:~#

kwindrem, 2022-11-18:
v4.29 of SetupHelper should not have these issues. With that version there are no "file sets". Rather, PageSettings.qml is modified in-line in the setup script making the package independent of Venus OS version.

Please update SetupHelper on your system.