kwindrem/SetupHelper #33: Guimods 7.4 don't work on Cerbo GX

Zabrickx, 2022-12-07:
Installing is no problem, but after restarting the graphical interface is not OK. I took a screenshot.
[image]
Cerbo GX work with firmware version 2.92 large.
Thanks for the interesting tools for VenusOS.
regards
Jean Pierre

kwindrem, 2022-12-07:
GuiMods v7.4 works fine on my Cerbo running v2.92 (normal and large).

Zabrickx, 2022-12-07:
Hello Kwindrem, thanks for your quick reply. 
All my devices, such as the four solar charge controllers, are connected with ve-direct. 
Could that have something to do with this? 
Greatings 
Jean-Pierre

Zabrickx, 2022-12-07:
Hello Kwindrem, after rebooting the Cerbo everything works fine. I thought I should just restart the gui. 
I'm glad everything works properly, I'm looking forward to additional options. Many thanks for this help. 
I wish you much success in all that you do. 
regards 
Jean Pierre

kwindrem, 2022-12-07:
I have seen cases where devices did not appear after a Venus OS update. Nothing at all to do with GuiMods. Reboot always brings things back. Should have mentioned a reboot.

closed 2022-12-07 by kwindrem