kwindrem/SetupHelper #35: 4.31 menu error not fixed

NickDolfin, 2022-12-24:
Sorry to inform you, but I still have a problem with the up/down arrows in ver 4.31
It hangs after a few clicks and the only escape is to press on pages.

kwindrem, 2022-12-24:
I'll look into it. It's only an issue when using the navigation buttons on a CCGX or the virtual equivalents on a remote console web page. It has to do with scrolling off the bottom of the available menu items. Tapping on the touchscreen seems to work fine.

NickDolfin, 2022-12-25:
Aha,

I have no touchscreen - only remote console in webbrowser

kwindrem, 2022-12-26:
There were some QML changes in Venus OS v3.00~14 that precipitated this issue and I have yet to find a cause. My code that worked in v2.92 (and probably v3.00~8) no longer works and the recommended change makes the cursor navigation of objects on several of my pages hang as you have discovered.

The problem is caused by cursoring to an object that is not visible. When this happens, the cursor controls  no longer function.

A workaround that may work for you is to click on the object rather than using the navigation buttons or the keyboard cursor keys.

kwindrem, 2023-01-16:
This issue is now fixed in SetupHelper v4.32

closed 2023-01-16 by kwindrem