kwindrem/SetupHelper #37: ExtTransferSwitch is missing from the package list

zorrobyte, 2023-01-02:
[image]

[image]

ExtTransferSwitch's readme says to use SetupHelper for install and provides no manual packages for USB install. However, ExtTransferSwitch is not listed as installable.

[image]

There is also an unpopulated "new" entry.

kwindrem, 2023-01-02:
Use new to enter the information for this package:
name: ExtTransferSwitdh
GitHub user: kwindrem
GitHub branch: latest

It will then appear in the Active packages list. With auto downloads and installs active, it will eventually automatically install, or you can manually download and install from within the package from the Active packages list.

zorrobyte, 2023-01-02:
The best, thanks!

closed 2023-01-02 by zorrobyte