kwindrem/SetupHelper #39: Junk files in the release tar.gz file

infamy, 2023-02-05:
You have a ton of ._ files in the release tar.gz. These are cache files. I would recommend moving the release process to something like github workflows/ci/cd to prevent this. (same thing is present in guimods)

kwindrem, 2023-02-05:
This was true at one time but I just checked the current release for SetupHelper and GuiMods and see no ._ files. The only thing I found was one .DS_Store file which I'll remove on the next release.

I also checked the blind install archive and found nothing there.

If you have a specific example, please let me know and I'll work on removing them.

infamy, 2023-02-06:
In the link in the readme you link to 

    venus-data.tgz is available here:
        https://github.com/kwindrem/SetupHelper/raw/main/venus-data.tgz

Inside that file, i see a ._rc and then each directory is littered full of ._folderorfilenamehere

kwindrem, 2023-02-06:
Got it. Didn't realize you were talking about the blind install which is different than what gets into the repo.

I have a fix and will include it in the next release. Unless you disagree, I don't consider this a serious issue requiring a new version just for this.

infamy, 2023-02-06:
Yeah, I don't think it causes any issues. Just extra files. if there a reason you don't build the blind release with github actions? (would just be an extra assset)

kwindrem, 2023-02-06:
I don't know that much about GitHub. It might be possible although I go outside the root file for SetupHelper to generate the necessary code since rcS.local and the rc directory needs to end up in /data not inside /data/SetupHelper.