kwindrem/SetupHelper #41: How to remove an active package from device via SetupHelper GUI?

TimD1981, 2023-02-24:
Hi,
I tried Venus OS V3.00-19 (version in developement) with SetupHelper 4.34 on a fresh installed environment with RaspberryPI (3).
The SetupHelper was installed successfully. Out of the Box in Setup Helper the setting "auto install" and "auto github download" are both disabled. I selected the Package RpiTemperature from the "inactive packages" list. After that, the package was displayed in "active packages" view. I couldn't see a button download or install, after clicking on the package RpiTemperature.  I had to change the settings for "auto install" and "auto download" to true. After some seconds the Package was downloaded an installed. Now I like to remove the active package. But I can't see a "uninstall" button.
[image]

kwindrem, 2023-02-24:
Thanks for the report. I'll look into it.

kwindrem, 2023-02-24:
Should be fixed now (v4.35)

closed 2023-02-24 by kwindrem