kwindrem/SetupHelper #44: Multiplus Grid Charging based on SOC or off-peak.

Regor48, 2023-03-26:
Is there a way to make the Cerbo GX to control the charging on the Multiplus, make it a charge only in off-peek hours (in my case m-f 7pm - 7am and weekends) and/or if SOC is low  and only to a defined SOC (so room left in battery to accept a solar charge). Possibly by toggling the setting on inverter details from "Inverting Only" to "On" then back to "Inverting only" at defined SOC.  

At this point I have a small system mainly to support critical loads (fridge, freezer, etc) and wonder if controlling the charging could reduce my energy consumption especially during peak hours without having to physically unplugging the charger during peak hours

kwindrem, 2023-03-26:
This is beyond the scope of any of my packages. This might be something you could accomplish with NodeRed but I know nothing about it.

Best post your question to the Victron forum.

closed 2023-03-26 by kwindrem

Regor48, 2023-03-26:
Thanks for the feedback and where to look. At least now with your package I
can turn off the charger remotely.

Roger