kwindrem/SetupHelper #45: Loading Guimods on v 2.92

rogermija, 2023-03-29:
Loading Guimods on Cerbo GX v 2.92 does not produce anything.  I tried previously to load on 2.6 but it load one time but did not provide full functionality.  Should any vestiges of Guidmods be removed to ensure a clean install?  If so, how would this be accomplished...through Cerbo USB port in terminal mode...or factory default?

kwindrem, 2023-03-29:
You might take a look at the SetupHelper log: /data/log/SetupHelper. If there were any GuiMods install failures they should appear there.

GuiMods has a file set for Venus OS v2.93. I don't know for sure if that will work for v2.92. An "incomplete file set" message should be in the SetupHelper log if it does not.

GuiMods install should override older installs. If you wanted to, you can run

/data/GuiMods/setup and choose uninstall

There are settings in / Settings / Display & language / GuiMods to enable and disable nearly every aspect of GuiMods so make sure the items you want to use are enabled.

I'm not sure GuiMods will install on Venus OS v2.6. The oldest file set is for v2.66 but that could work for earlier versions also.

rogermija, 2023-03-29:
Thank you for your response.  The amount of work you have put forth is
exceptional.