kwindrem/SetupHelper #47: Venus OS blind install mechanism

mr-manuel, 2023-05-08:
Hello Kevin,

I'm searching how the automated install mechanism of Venus OS drivers works. Like you put the `venus-data.tgz` on a USB drive, insert the drive and Venus OS will install it.

Maybe you have a link to a documentation or something?

If not, maybe you could help me with a few questions?

1. Where is the script in Venus OS located, that is called if you insert a USB drive with a `venus-data.tgz`?
2. Is there a specific folder structure that has to be preserved in the tar?
3. Which script is called from Venus OS for install after the extraction of the `venus-data.tgz`? I have the sensation that it searches only for some file with `local` in the name and executes it.

Thanks a lot!

mr-manuel, 2023-05-08:
As it is so often the case, you have to ask someone else to find it for yourself, else you will not find it :-)

This answers all questions:
https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-core/initscripts/files/update-data.sh

closed 2023-05-08 by mr-manuel

kwindrem, 2023-05-08:
update-data.sh unpacks the archive to /data.
There are also pre and post scripts wrapped around the unpack. These must be in  the rc folder inside the archive.

The restore to factory defaults has one file in the archive: rcS.local. That script is executed during boot and simply removes (almost) everything in /data.

Look at the blind install scripts in SetupHelper for more ideas.