kwindrem/SetupHelper #50: Venus-data.tgz download

dust900, 2023-05-31:
Hi, the file that downloads via the link for a blind install has a .tar extension.  Not .tgz.  Loading this file into the cerbo....nothing changes when I reboot.  Firmware 3.00.  This is the first time I am trying to load guimods.

kwindrem, 2023-05-31:
The system used to download the package or the blind install file may be unzipping the file. Safari does this by default and you need to turn off the option: Open "safe" files after downloading. in the General preferences.

I did notice that the blind uninstall file is .tar.gz which is OK as is. I'll change that to .tgz the next time I update SetupHelper for consistency.

dust900, 2023-05-31:
Thx for the response.  The folder/file is not unzipped when I check the download or the file on the USB.

￼

Unzipped:

￼

kwindrem, 2023-05-31:
Just downloaded venus-data.tgz from GitHub and ended up with exactly that file name.

If you place that file on the USB stick the blind install should work.

You can not unzip the file which ends up with a .tar file. That won't be recognized by Venus OS.