kwindrem/SetupHelper #54: Remove package should also remove from filesystem?

mr-manuel, 2023-06-29:
Should the "Remove package" button also delete the package from the filesystem or only remove it from the active packages?

If it should only remove it from the active packages I would at least ask, if the package should also be removed from the filesystem.

kwindrem, 2023-06-29:
Remove package flags the package so it will not be automatically reinstalled. With this flag set, the package just sits in /data. If you want to completely remove it, you'll need an ssh connection or local terminal.

kwindrem, 2023-06-29:
When a package is removed, it moves from Active packages to Inactive packages. It never really disappears from the GUI but it isn't shown as an Active package any longer.

I wanted to avoid lots of interaction during the processes managed from the GUI using PackageManager. It really is not a substitute for ssh interaction, rather something unskilled users can wrap their head around.

What I was trying to avoid by not deleting the package from /data was a download again if the user wanted to add the package again. There are systems that don't have a robust internet connection.