kwindrem/SetupHelper #55: Issues in v3.20~18 or slightly before

mpvader, 2023-11-07:
freespace on a Cerbo v3.12~1:

```
root@einstein:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               320.3M    242.6M     56.2M  81% /
devtmpfs                454.5M      4.0K    454.5M   0% /dev
tmpfs                   503.0M    904.0K    502.1M   0% /run
tmpfs                   503.0M    284.0K    502.7M   0% /var/volatile
/dev/mmcblk1p5            1.1G    123.0M    955.7M  11% /data
tmpfs                   503.0M    904.0K    502.1M   0% /service
overlay                 503.0M    284.0K    502.7M   0% /var/lib
```

freespace after installing v3.20~18 large, before resize2fs.sh
```
root@einstein:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root              1006.3M    997.2M         0 100% /
devtmpfs                454.5M      4.0K    454.5M   0% /dev
tmpfs                   503.0M    948.0K    502.0M   0% /run
tmpfs                   503.0M    228.0K    502.7M   0% /var/volatile
/dev/mmcblk1p5            1.1G    123.1M    955.7M  11% /data
tmpfs                   503.0M    948.0K    502.0M   0% /service
overlay                 503.0M    228.0K    502.7M   0% /var/lib
root@einstein:~# 
```

mpvader, 2023-11-07:
After above it ends up being in a reboot loop every few minutes (some watchdog I expect), and the GX Touch shows totally black except for a short blip

```
U-Boot SPL 2018.05 (Apr 12 2023 - 23:29:03 +0000)
DRAM: 1024 MiB
CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC2

U-Boot 2018.05 (Apr 12 2023 - 23:29:03 +0000) Allwinner Technology

CPU:   Allwinner A20 (SUN7I)
Model: Cubietech Cubieboard2
I2C:   ready
DRAM:  1 GiB
MMC:   SUNXI SD/MMC: 0, SUNXI SD/MMC: 1
Loading Environment from MMC... OK
In:    serial@01c28000
Out:   serial@01c28000
Err:   serial@01c28000
Net:   No ethernet found.
Setting bus to 1
Hit any key to stop autoboot:  0
30726 bytes read in 8 ms (3.7 MiB/s)
3825376 bytes read in 222 ms (16.4 MiB/s)
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
   Loading Device Tree to 49ff5000, end 49fff805 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.109-venus-12 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 9.5.0, GNU ld (GNU Binutils) 2.34.0.20200910) #1 SMP Sun Nov 5 22:57:12 UTC 2023
[    0.000000] CPU: ARMv7 Processor [410fc074] revision 4 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Cerbo GX
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x4a000000, size 96 MiB
[    0.000000] OF: reserved mem: initialized node default-pool, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]
[    0.000000]   HighMem  [mem 0x0000000070000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 15 pages/cpu s31564 r8192 d21684 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mmcblk1p3 rootwait ro rootfstype=ext4 vt.global_cursor_default=0
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 930756K/1048576K available (6144K kernel code, 647K rwdata, 1304K rodata, 1024K init, 221K bss, 19516K reserved, 98304K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x314/0x4ac with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000023] Switching to timer-based delay loop, resolution 41ns
[    0.000469] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000812] clocksource: hstimer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370868154 ns
[... 359 lines cut ...]
ls: /data/var/lib/connman: No such file or directory
starting IPv4LL configuration daemon: avahi-autoipd... Cannot find device "eth0"
start-stop-daemon: unable to close pidfile '/var/run/avahi-autoipd.pid' (No space left on device)
start-stop-daemon: child returned error exit status 2
done.
starting resolv.conf manager: resolv-watch... start-stop-daemon: unable to close pidfile '/var/run/resolv-watch.pid' (No space left on device)
start-stop-daemon: child returned error exit status 2
done.
Starting crond: OK
cat: can't open '/tmp/last_boot_type': No such file or directory
Checking available software versions...
Active rootfs: 2
tee: /var/run/versions: I/O error
Active version: 20231105204412 v3.20~18
[   30.536394] EXT4-fs (mmcblk1p2): mounted filesystem with ordered data mode. Opts: (null)
tee: /var/run/versions: I/O error
Backup version: 20231027124454 v3.12~1

Victron Energy v3.20~18 einstein ttyS0

root@einstein:~#

U-Boot SPL 2018.05 (Apr 12 2023 - 23:29:03 +0000)
DRAM: 1024 MiB
```

mpvader, 2023-11-07:
this probably doesn't help either:

```root@einstein:~# cat /etc/fstab
root@einstein:~#```

kwindrem, 2023-11-08:
SetupHelper v5.18 has been released that should fix this issue. With this release, the root partition is resized (I thought I was doing this already but was NOT so now it's fixed). I also check the root partition for space to install a package before doing so.

The issue of nonfunctioning systems still remains. Matthijs has provided some instructions in https://github.com/kwindrem/SetupHelper/issues/56 but in many cases you will need to reimage the system. For those on a Raspberry PI you may be able to switch to the alternate root FS using the boot loader:

editenv version
change 1 to 2 or 2 to 1

saveenv

boot

closed 2023-11-08 by kwindrem