kwindrem/SetupHelper #56: How to reflash a Cerbo GX

mpvader, 2023-11-07:
I've (quickly) made these notes to help people that ran in to trouble when using v3.20~16 or later in combination with Guimods or SetupHelper.

Note that this will reset the device to its defaults, all settings need to be redone. Also on VRM you'll need to reset the authentication code.

And, on somewhat older Cerbos, this will break the onboard wifi access point.

### 1. Download a recent install image here:

https://updates.victronenergy.com/feeds/venus/release/images/einstein/

Note that files in that folder are for the Cerbo GX. For other devices, go one dir up and use the right folder.

For Cerbo-S GX, use the cerbosgx folder.

For other devices, use the other folders, but knowing bit what is going on here, I expect/hope that really only the Cerbo and Cerbo S are affected.

### 2. Install the image on a usb stick or sdcard

I use RaspberryPi imager for that. Find it in Google. The option to select your own file is at the bottom of the list, see here:

[image]

Note that there is no need to unzip the zip file.

Once done, you'll see this:

[image]

### 3. Installing the image

Remove the sdcard from the computer, insert it in the Cerbo and power cycle it.

mpvader, 2023-11-07:
Nobody normally has access to a serial console, but this is what you get to see there if you were looking:

```
Searching for installer image...
Installer image found on mmcblk0p1
Creating partitions...
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk1: 3.7 GiB, 3959422976 bytes, 7733248 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x564e5553.
/dev/mmcblk1p1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 8 MiB.
/dev/mmcblk1p2: Created a new partition 2 of type 'Linux' and of size 1.3 GiB.
Partition #2 contains a ext4 signature.
The signature will be removed by a write command.
/dev/mmcblk1p3: Created a new partition 3 of type 'Linux' and of size 1.3 GiB.
Partition #3 contains a ext4 signature.
The signature will be removed by a write command.
/dev/mmcblk1p4: Created a new partition 4 of type 'Extended' and of size 1.2 GiB.
/dev/mmcblk1p5: Created a new partition 5 of type 'Linux' and of size 1.2 GiB.
Partition #5 contains a ext4 signature.
The signature will be removed by a write command.
/dev/mmcblk1p6: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x564e5553

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk1p1 *       2048   18431   16384    8M  c W95 FAT32 (LBA)
/dev/mmcblk1p2        18432 2639871 2621440  1.3G 83 Linux
/dev/mmcblk1p3      2639872 5261311 2621440  1.3G 83 Linux
/dev/mmcblk1p4      5261312 7733247 2471936  1.2G  5 Extended
/dev/mmcblk1p5      5263360 7733247 2469888  1.2G 83 Linux

Filesystem/RAID signature on partition 2 will be wiped.
Filesystem/RAID signature on partition 3 will be wiped.
Filesystem/RAID signature on partition 5 will be wiped.

The partition table has been altered.
Calling ioctl() to re-read partition table.
[    6.694578]  mmcblk1: p1 p2 p3 p4 < p5 >
Syncing disks.
Formatting data partition...
mke2fs 1.45.7 (28-Jan-2021)
Discarding device blocks: done
Creating filesystem with 308736 4k blocks and 77280 inodes
Filesystem UUID: 2ded2fe2-48bf-43cd-a8c0-e88dc27171e4
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): [   12.527053] random: crng init done
done
Writing superblocks and filesystem accounting information: done

Creating /data/venus/installer-version...
[   13.732477] EXT4-fs (mmcblk1p5): mounted filesystem with ordered data mode. Opts: (null)
Installing rootfs1...
Swupdate v2016.10.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Registered handlers:
        raw
        rawfile
        uboot
software set: stable mode: copy1
Main loop Daemon
Version 0.1.0
Warning: Bad CRC, using default environment
Software updated successfully
Please reboot the device to start the new software
[NOTIFY] : SWUPDATE successful !
Installing bootloader...
Swupdate v2016.10.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Registered handlers:
        raw
        rawfile
        uboot
software set: stable mode: bootloader
Main loop Daemon
Version 0.1.0
Software updated successfully
Please reboot the device to start the new software
[NOTIFY] : SWUPDATE successful !
Installing firmware...
Open On-Chip Debugger 0.10.0+dev-00954-gded679902-dirty (2023-10-30-22:52)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : SysfsGPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : This adapter doesn't support configurable speed
Info : SWD DPIDR 0x0bc11477
Info : at91samc21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : at91samc21g18.cpu: external reset detected
Error: couldn't bind gdb to socket on port 3333: Cannot assign requested address
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* at91samc21g18.cpu  cortex_m   little at91samc21g18.cpu  running

Info : SWD DPIDR 0x0bc11477
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x000042c0 psp: 0x20000ac0
** Programming Started **
Info : SAMD MCU: SAMC21G17A (128KB Flash, 16KB RAM)
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : SWD DPIDR 0x0bc11477
shutdown command invoked
Installation complete
Remove installer medium and power cycle system
```

mpvader, 2023-11-07:
Wrt restoring the onboard access point:

that is failing because a few required files are missing. Only on older Cerbos. 

The most user friendly solution that I can think about, and which doesn't require logging into ssh, would be see if a a venus-data.tgz can be made that creates the required serial number + wifi code files on the data partition. And, since this problem and all including guimods is a community effort: anyone out there that can work on that is welcome too!)

AlbertronicDev, 2023-11-08:
@mpvader Hi Matthijs, is it possible that VenusOS increases the root partition before installing the beta? Or maybe a intermediary beta update which does specifically this, so that the root partition is increased before updating to the "real" latest beta.

I see @kwindrem has updated SetupHelper so that the root partition is resized/increased, but many users do not have automatic updates activated on SetupHelper/PackageManager. This means that updating VenusOS to the latest beta is possibly going to fail at a lot of SetupHelper/PackageManager users.