kwindrem/SetupHelper #57: Failed installation of V3.20~18 with setuphelper 5.11

EJhagenaar, 2023-11-08:
I have a Cerbo S GX. Before the update to V3.20-18 failed. I recovered the firmware with a SD card to V3.12 and installed the latest setuphelper 5.11. I continued installing gui mods, BatteryAggregator, dbus-mqtt-devices and serialbattery.
When this was finished i updated the firmware to 3.20~18 and i have the same as before. (white insert in the console)
[image]

kwindrem, 2023-11-08:
Also, please install one package at a time as there may be conflicts. This may help narrow down which package is causing the white screen.

The GUI log will also point to a specific file at fault:

/data/log/gui/current is the GUI log file

kwindrem, 2023-11-08:
Also please specify if you are installing the normal or large version.

EJhagenaar, 2023-11-14:
I am running the large image. Cerbo S GX. but as the ssh password can not be set after the update (unresponsive UI) i can not ssh into the device and check anything.

kwindrem, 2023-11-14:
The only thing I can recommend is that you reimage again then upgrade to v3.20~23 (or whatever is the latest beta) then set up ssh access from the local GUI. Only then try installing one package at a time.

EJhagenaar, 2023-11-17:
I did that and that worked.

closed 2023-11-17 by EJhagenaar