kwindrem/SetupHelper #58: Won’t run with v3.20~27

testpaul999, 2023-11-21:
Hi, 
After Upgrade Venus os to ~27 the PackageManager disappear from the Victron Web-UI.
Go back to ~26 all fine.
Regards

EngMarc, 2023-11-21:
I just tried to install SetupHelper on a new Ekrano GX with firmware 3.12. It didn't load the USB  but the USB drive did register and light up and it used it for logging. I tried both versions of SetupHelper (the one released today and the prior one). Neither load into the Ekrano menus.

Boot103, 2023-11-21:
See closed issue https://github.com/kwindrem/SetupHelper/issues/59

kwindrem, 2023-11-21:
EngMarc, make sure SetupHelper v5.15 is installed. V5.14 WILL have the problem you describe.

if you are having problems with SetupHelper v5.15, please provide additional information like logs.

SetupHelper modifies /opt/victronenergy/gui/qml/PageSettings.qml. There should be a line at the top of this file indicating it is the modified version. There should also be a PageSettings.qml.orig in the same directory.

The reason going back to a previous version using the backup firmware option is that SetupHelper is already installed there so doesn't need to be reinstalled.

EngMarc, 2023-11-21:
where do I get the logs? 
I'm new to the Ekrano GX and trying to get things set up.
Also, should I unzip the SetupHelper-main.zip file when I put it on the USB?
The people out there say it doesn't matter if it is zipped or unzipped.

kwindrem, 2023-11-21:
If you are trying to use the blind install process, you have the wrong file. You need to download the package to a computer then unzip it. In the folder that is created, you will find a venus-data.tgz file. Put that on the USB stick (do not unzip it -- it DOES matter !!!), then put that into your Eranko.

You'll find more instructions in the ReadMe file inside the file you unzipped above. or available here:

https://www.dropbox.com/scl/fo/bx5aftvgrqq0vp060mwip/h?rlkey=k28c2i49fjfpcyjfsuldwp159&dl=0

Logs are in /data/logs. The files of interest would be /data/log/SetupHelper

ssh access is described here:

https://www.victronenergy.com/live/ccgx:root_access

closed 2023-11-21 by kwindrem