kwindrem/SetupHelper #59: Package Manager Not Appearing V3.12 Build 20231030212506

Boot103, 2023-11-21:
Hi, 

Im trying to install SetupHelper on my new CerboGX

I have tried the USB Stick method and command line via putty. 

When I reboot I never get the package manager appear in the settings menu. Any ideas?

Logs attached. 

[SetupHelper.txt](https://github.com/kwindrem/SetupHelper/files/13427167/SetupHelper.txt)

Line 56 and 57 stand out but this is about the limit of my knowledge on this stuff

@40000000655bfdb008a35b6c SetupHelper/setup: ERROR: no soure file for replacement /opt/victronenergy/gui/qml/PageSettingsPackageEdit.qml - can't continue with install
@40000000655bfdb0138b5584 SetupHelper/setup: ERROR: no soure file for replacement /opt/victronenergy/gui/qml/MbDisplayPackageVersion.qml - can't continue with install

Boot103, 2023-11-21:
Re imaged to latest release candidate https://updates.victronenergy.com/feeds/venus/candidate/images/einstein/

Formatted SD card and moved latest venus-data.tgz onto the SD card. If the SD card is inserted and I reboot the cerbo GX it just has dim red lights where the bluetooth and wifi should be. 

If I eject the SD card it boots fine?

kwindrem, 2023-11-21:
v5.15 fixes PackageManager not showing up

Not sure about the other errors reported above. Let me know it you have further problems with blind install

closed 2023-11-21 by kwindrem