kwindrem/SetupHelper #61: setuphelper missing in venus os v3.20~33

thomasinaz41213, 2023-12-11:
after doing a fresh install of venus os v3.20~33 the package manager disappeared from settings. I waited several minutes and then reboot to no avail. Rolled back to v5.15 of setup helper and package manager came back. I tried switching back to venus os v3.13 and doing a repeated fresh install of beta v3.20~33 but the only way I could get package manager back in the settings menu was to revert to v5.15 of setup helper.

kwindrem, 2023-12-11:
How did you do the fresh install of v3.20~33? on-line update or SD card?

Can you get to logs? Use the backup function in Package Manage and you can copy all log files to an SD card then post the .zip file here.

I am switching between firmware versions all the time and have not seen this problem.

thomasinaz41213, 2023-12-11:
I did an online update from 3.13 to 3.20~33. I didn't really need to but just wanted a clean slate since I had been playing with some bms serial drivers. Since nobody else has mentioned an issue it may have just been a glitch with my system. It happens sometimes. Such is life when you mess around under the hood.

kwindrem, 2023-12-11:
FYI, SetupHelper is now at v5.16. That version fixed a bug that could result in a white screen when on the first boot after a Venus OS upgrade. A second reboot showed the GUI again.

v5.15 was supposed to fixe a problem where PackageManager didn't show in the menus like you were seeing. So I wonder if somehow you were not on v5.15 or maybe latest was actually v5.14. I verified that latest is actually v5.16 and also that the blind install is v5.16 so hopefully we're beyond your problem. I'll close for now but feel free to reopen if you see it again.

closed 2023-12-11 by kwindrem

thomasinaz41213, 2023-12-17:
Have the same problem again updating from v3.13 to v3.20~34. Got the following from the log....same comment for each package attempting to be re-installed.
@40000000657e41a42ea903d4 SetupHelper/setup: no room for modified files on /root (0 MB remaining) - can't continue

kwindrem, 2023-12-17:
What version SetupHelper?

There was a bug a while back that failed to resize the root partition. That was fixed in v5.11

Please check to see if /root is actually mounted rw and resized and how much free space actually exists.

kwindrem, 2023-12-17:
I remember making a change recently to avoid the nasty message of remounting/resizing on each run of any setup script. I think that change may prevent a system with /root mounted read-write but not resized from skipping the resize step.

For now you can manually run /opt/victronenergy/swupdate-scripts/resize2fs.sh.

thomasinaz41213, 2023-12-17:
Setuphelper is at v5.16. As far as your other questions, can you tell me how to find that information?

kwindrem, 2023-12-17:
You need ssh access. Then: 

df -m / 

will give you free space in MB.

To determine if root is mouted read-write:

mount

root ( / ) should be the first entry

thomasinaz41213, 2023-12-17:
I resized as you instructed above and everything re-installed properly after that.

kwindrem, 2023-12-17:
Great. I'll fix the remount but not resize hole and push a new version.

kwindrem, 2023-12-17:
SetupHelper v5.17 is available that fixes the remount but not resized bug.