kwindrem/SetupHelper #63: no room on root partition

stepir, 2023-12-29:
while trying to upgrade on a CCGX

[image]

however:

```
root@ccgx:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
ubi0:rootfs2            190680    124580     66100  65% /
devtmpfs                113496         4    113492   0% /dev
tmpfs                   122200       820    121380   1% /run
tmpfs                   122200       568    121632   0% /var/volatile
ubi0:data                29032      9048     18488  33% /data
tmpfs                   122200       820    121380   1% /service
```

kwindrem, 2023-12-29:
Can you provide the SetupHelper log?

Seems like there should be room.

What version of SetupHelper are you running.

Note the checks I added recently are to avoid bricking a GX device that did not have room to install packages so they are really important. The code fails an install if there is not 10 MB available. It looks like your system has 33 MB available (or maybe it's 66 MB) on root.

df -m will report the used/avaiable space n MB/

johnyhravy, 2024-03-06:
Hello.

I'm using SetupHelper on Raspberry Pi4 with Venus OS and last year around August 2023, after updating the Venus firmware the system crashed and wouldn't boot and I had to connect the Rpi with a LAN cable to the NTB and restore the firmware from the backup firmware repository.
I don't know if it's related to this, but my SetupHelper doesn't update either. The last packages I had were GuiMods v8.10 (act.v10.0), SetupHelper v5.6 (act.v6.3), it doesn't work even if I manually download a newer version.
No idea where the error might be?
Later, the Package manager shows me as not running and I can't install either as a "blind install" or from the command line.
I read something about the installation files being changed in new versions of fw. or I'm also thinking about the USB flash error. I tried to read it in NTB and it reported a media error, but even after fixing it, nothing changed.

Later I read other discussions and it is probably full Rpi memory.
I am attaching a list of currently and previously used mods.

current applications:
Shelly EM
RpiTemperature
SerialBattery

previously used:
BatteryAggregator
GeneratorConnector
GuiMods
RpiGpioSetup
SetupHelper
ShutdownMonitor

TRYING TO INSTALL SETUPHELPER
root@raspberrypi4:~# wget -qO - https://github.com/kwindrem/SetupHelper/archive/   latest.tar.gz | tar -xzf - -C /data
tar: SetupHelper-latest: Cannot mkdir: No space left on device
tar: SetupHelper-latest: Cannot mkdir: No space left on device
tar: SetupHelper-latest/.github: Cannot mkdir: No such file or directory
tar: SetupHelper-latest: Cannot mkdir: No space left on device
tar: SetupHelper-latest/.github/workflows: Cannot mkdir: No such file or directo   ry

root@raspberrypi4:~# df -m
Filesystem               1M-blocks      Used   Available Use%   Mounted on
/dev/root                          1040         989                 0   100%   /
devtmpfs                         1886              0          1886       0%   /dev
tmpfs                               1919              1          1918       0%   /run
tmpfs                               1919              0          1918       0%   /var/volatile
/dev/mmcblk0p1                 41            22              19      53%  /u-boot
/dev/mmcblk0p4           20305      20289                0   100%   /data
/dev/sda1                        7643              0           7643       0%   /run/media/sda1
tmpfs                               1919              1           1918       0%   /service

Don't know which can be filling memory and how to clear it and restore SetupHelper?

Thanks for the advice.

johnyhravy, 2024-03-06:
I'm sorry, the memory table was scattered, I'm attaching it as an attachment.
[image]

kwindrem, 2024-03-06:
A full file system probably explains your issues.

closed 2024-03-06 by kwindrem

johnyhravy, 2024-03-06:
Thank you for your quick reply. Can you please advise me how to restore the functionality of the system? I hope it won't be a full reinstall of Venus OS...

kwindrem, 2024-03-06:
You need to figure out what filled up the /root and /data partitions and remove it. Sorry, I can't help you do that because I have no idea what might have caused it. I'd go down one directory level at a time using du to hopefully find the fault.

You could also put the SD card in a system that can evaluate/repair ext4 file systems to see if repairing it will help. Or maybe run fsck on the GX device.