kwindrem/SetupHelper #65: ERROR: can't remount root read-write - can't continue

trebeljahr, 2024-01-03:
I logged into the ccgx device via Shell. And executed the commands from the ~ directory in sequence, downloading, unpacking, moving and executing the start scripts because it wasn't working with the blind install method somehow... However on the last command running the script I get an error and the script fails (which explains why the blindInstall didn't work)

Commands run: 
```
root@ccgx:~# wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C /data
root@ccgx:~# mv /data/SetupHelper-latest /data/SetupHelper
root@ccgx:~# /data/SetupHelper/setup
```

Output of the script: 
```
--- starting setup script v5.17
remounting root read-write and resizing
Rootfs is not ext4, but ubifs
ERROR: can't remount root read-write - can't continue
```

Also in case it helps here are the outputs of the df and df-m commands: 
```
root@ccgx:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
ubi0:rootfs2            190680    136520     54160  72% /
devtmpfs                113496         4    113492   0% /dev
tmpfs                   122200       752    121448   1% /run
tmpfs                   122200       276    121924   0% /var/volatile
ubi0:data                29032      3180     24356  12% /data
/dev/sda1              3903000       980   3902020   0% /run/media/sda1
tmpfs                   122200       752    121448   1% /service
root@ccgx:~# df -m
Filesystem           1M-blocks      Used Available Use% Mounted on
ubi0:rootfs2               186       133        53  72% /
devtmpfs                   111         0       111   0% /dev
tmpfs                      119         1       119   1% /run
tmpfs                      119         0       119   0% /var/volatile
ubi0:data                   28         3        24  12% /data
/dev/sda1                 3812         1      3811   0% /run/media/sda1
tmpfs                      119         1       119   1% /service
``` 

To me, I am good with software development but have never worked with any of the Venus OS / CCGX battery management stuff, so if there is something that I clearly did wrong/should have done or configured differently, please let me know :) 

But as it stands now the script is broken for me. If you need, I can also provide you with more information to help debugging, just let me know. 

Also, after some tracing, the line in the script that is failing, is not in the scripts in this repo but somewhere on the firmware itself: 

Specifically over here: 
/opt/victronenergy/swupdate-scripts/resize2fs.sh

Which is a line executed from the SetupHelper-blind/CommonResources bash script.
 
The code that is breaking in the resize2fs on the machine looks like this: 
```
fs=$(grep -E '^\S+\s+/\s' /proc/mounts | awk '{print $3}')
if [ "$fs" == "ext4" ]; then
    echo "resizing $root"
    resize2fs $root
else
    echo "Rootfs is not ext4, but $fs"
fi
```

The software version of the OS on the CCGX should be v3.20~15, we updated it just before running all of this.

kwindrem, 2024-01-03:
Working on it now.

This issue was created in v5.17 with a change to make other platforms resize the root partition properly. The root partition on my CCGX had already been mounted read-write so I didn't encounter the problem.

For now run the remount-rw.sh script manually. Then you should be able to run the setup script again.

kwindrem, 2024-01-03:
fixed in v5.18

closed 2024-01-03 by kwindrem