kwindrem/SetupHelper #67: Package install stuck in loop

thomasinaz41213, 2024-02-01:
I have a few packages that I have installed using package manager. One is a modified fork of RemoteGPIO that I have modified for tcpip use, and the other is a simple script to enable node red context storage. I have had no issues with these packages during several Venus os updates. Upgrading from v3.20-40 to v3.20-43 these 2 packages got stuck in a loop (I'm assuming because when I looked in the package manager menu the status kept rolling from checking for downloads to installing these 2 packages) but the packages never got installed. I checked the FileSets directory for each package, and setup helper had successfully created v3.20-43 fileset for each package with the correct replacement files and created a COMPLETE file in each directory. (There were no changes in the files to be modified between v3.20-40 and v3.20-43) Rolling back to v5.18 of setup helper and the packages successfully installed and are working as expected. Is there a change or a flag I need to add to the packages for the latest version of setup helper to work with my packages? I am aware that there were some recent changes as to how packages are handled.

kwindrem, 2024-02-01:
I downloaded your RemoteGPIO_tcpip package and tried to install it and the file set failed with

I show this failing with: service  not found - can't continue

Your setup script has an installService call but there is no service in the package directory.

I think the reason this succeeds in SetupHelerp prior to v6.0 is that one of the changes I made in v6.0 was to prevent install if any failure occurs during installation. I don't think that existed in previous versions. You should update your setup script to remove the installService and removeService lines since your package does not appear to have any services.

I did reproduce the repeated install you are seeing and will look at a fix for that.

BTW, this package apparently only works on Cerbo or maybe just not on Raspberry PIs since the gpio_list stock file differs from what is in your file sets.

thomasinaz41213, 2024-02-01:
Yes this package only works with cerbo not rpi. The service was part of fguiot original package but is not needed for tcpip so is not included in the package download. I completely overlooked that in the setup script. I have revised the setup script and pushed a new release and all works as expected now. Thanks for taking the time to check that out.

I suspect I probably have a similar issue with the context storage script so I will look at that as well.

kwindrem, 2024-02-01:
I've pushed SetupHelper v6.2 which fixes the repeated install on failure you saw.

closed 2024-02-01 by kwindrem