kwindrem/SetupHelper #71: Auto install not functioning

thomasinaz41213, 2024-03-04:
V6.9 auto install seems to have stopped working. When adding a package to the active list it will auto download, but not auto install. I have tried rebooting the system but it still won't auto install. Manual install works fine. I have tested with multiple packages. Verified on my production system as well as my test system.

kwindrem, 2024-03-04:
Note that if you manually uninstall a package it will not reinstall. The flag that controls this persists when you remove the package so even after you add the package back in it will not auto install. This may explain your situation. 

Let me know if this explains the behavior you experienced. If so, I may be able to address it by clearing the block on auto installs when adding a package.

kwindrem, 2024-03-04:
I went ahead and unblocked downloads and installs when adding a package. v6.10 is out now.

Reopen this issue if this did not solve your problem.

And thanks for contributing. It helps.

closed 2024-03-04 by kwindrem

thomasinaz41213, 2024-03-04:
So can you clarify something for me that is actually related to this thread?

On my test system I install and test various packages at different times, some of which can be variations of other packages, which would be conflicting. Am I to understand that I can uninstall a package but leave it in the active packages list, add a conflicting package and install it, and the uninstalled package will not attempt a reinstall even if the package has an update, or the system reboots?

I have been removing the package and then adding/installing the conflicting package. This makes for a lot of extra unnecessary steps if the above is true.

kwindrem, 2024-03-04:
Yes, what you say is true. If you manually uninstall a package, it won't auto install until you manually install it again. This is true even if the package you uninstalled updates from GitHub or removable media.

The change I just made WILL allow auto installs again if you remove the package and add it again using PackageManager.

I am also working on a conflict resolution mechanism. The first step is a way for one package to force uninstall another. This will be in v6.12 out shortly. Eventually, my plan is to identify package conflicts as automatically as possible then prioritize one package over another. Still working out how this might happen.