kwindrem/SetupHelper #72: conflict with multiple packages that modify/replace the same system files

thomasinaz41213, 2024-03-05:
this is in regards to the work currently being done on RemoteGPIO, and I assume any other packages that attempt to modify a previously modified file will have the same issue.

I have discovered that when setup helper attempts to install RemoteGPIO and current files do not have a .orig match in the filesets, the installation fails (as it should) but setup helper is then restoring the victron original file and not the file that was previously installed via the guimod package. For example, I forked the repo (RemoteGPIO) and added the missing services.json and services.json.orig files into a filesets folder for v3.30~11. installing the package fails because the guimod installed file MbItemDigitalInput.qml does not have a match in RemoteGPIO filesets. Setup helper then restored the original file and not the guimods file, thereby removing the additional entry for the ExtTransferSwitch Digital Input. The original victron MbItemDigitalInput.qml is then placed into the missing filesets folder as the .orig file and not the guimods version of MbItemDigitalInput.qml.

it then becomes necessary to uninstall and reinstall guimods to get the correct file back into victron.
if you want to have a look you can add my fork RemoteGPIO thomasinaz41213 v3.1.3

kwindrem, 2024-03-06:
Yes, I am aware of this type of conflict. Right now it's up to package authors to prevent the problem.

Restoring another package's modded file won't be an easy change. This is especially true because I can not determine the order which packages install.

I'm working on a mechanism where package authors can identify conflicts that will prevent one package installing over another but it still relies on the knowledge of the conflict and collaboration between the authors. But it's a step.

I think for RemoteGPIO, the package will require GuiMods and not modify files GuiMods changes further. The file you have identified may have slipped through the crack.

kwindrem, 2024-03-06:
Just checked and the MbItemDigitalInput.qml changes for v3.22 are known and will be incorporated into GuiMods and RemoteGPIO will not install this file.

kwindrem, 2024-03-06:
I am implementing a mechanism that should prevent one package installing over another.

I store the package name that actually updates a file. If not, the update  and the entire instal fails. Failure causes that package to uninstall. So the update from the first package is preserved.

This will be part of SetupHelper v6.12.

Note that packages installed with previous versions will not set the package info so there can be no conflict checks. But it should work going forward.

closed 2024-03-25 by drtinaz