kwindrem/SetupHelper #75: Using patched files

thomasinaz41213, 2024-03-13:
Kevin I have been looking at the documentation for the recent changes to setuphelper and am interested in using the patch function going forward. How would this effect previous versions of the package? Would I need to rebuild the package including previous venus firmware versions? Would updatePackage have a stroke and blow up any previous FileSets?

thomasinaz41213, 2024-03-13:
I'm going to give this a go with one of my simpler packages that only replaces one file.

So if I understand correctly, the files that need to be placed in the PatchSource directory are 'somefile.source' (a version of the active or original file) and 'somefile.edited' ( a version of the expected replacement file)? And then when I run updatePackage I will get a prompt to create the ,patch file, which will then also be placed in the PatchSource directory? Or is the .patch file then placed in the specific version fileset? And then I obviously also need to make the other necessary changes to the setup script. Do I have that correct?

kwindrem, 2024-03-13:
Exactly. You would also need to add the path to the active file to the fileListPatched file. This is what updatePackge uses to identify patch file creation.

updatePackage will also apply the patch file to each stock file in all fileSets and report if patch reports an error. Note that this isn't necessarily an exhaustive test since doing that would need the expected result for each version which defeats the purpose of patch-based replacements.

thomasinaz41213, 2024-03-14:
Kevin I am having an issue with one of the patch files in one of my packages. The first package I rebuilt for using patch files went pretty smooth, as it only patches a single file, and does nothing else. It works great.

Another package I am currently working on patches three files, one of which is fairly large, and is patched in two places (the latter being the problem file). The .patch file (in the PatchFile folder) has been created properly, and I have opened it and verified that the correct added lines of code do exist. The issue is that when setup runs and the active file is patched, only the first part of the patch is being applied, not the second part. Is this a known issue or limitation when patching files? Is there some other step I am missing?

kwindrem, 2024-03-14:
I know of no limitations. Have you tried manually applying the patch?

Is the file you are patching the same as the .source file or is it from another Venus OS version?

If it's from a different Venus OS version than the one you created the .patch file from, try running that manually. See if any errors are reported or any reject files created. Might provide a clue.

The BusyBox version of patch is a bit limited so you might transfer the source you are having trouble with and the .patch file to another machine and run it there with verbose enabled.

thomasinaz41213, 2024-03-14:
It turns out that the second chunk was being applied, but to the wrong part of the code. I simply re-created the patch file using diff -U6 to include 6 lines before and after the added lines, instead of the default 3. Works now. Thanks for your help.

closed 2024-03-14 by thomasinaz41213