kwindrem/SetupHelper #78: Setuphelper v7.0 release active packages malfunction

thomasinaz41213, 2024-03-15:
After a failed install of a package, clicking on any package in the active packages list results in the same screen. Only way to fix is either a system reboot or reverting setuphelper to an older version. Re-created on two systems.

thomasinaz41213, 2024-03-16:
[image]

My bad...forgot the picture. Clicking on any package in the active package list gives this screen.

kwindrem, 2024-03-16:
Is it the message that does not go away or is it that it stays on SetupHelper?

What do Previous/Next do if anything?

If you exit out to the Active packages menu and go back in to another package is the menu for the selected package or always SetupHelper?

BTW, you can now restart PackageManager (from the end of the main Package Manager menu. You can restart the GUI from there too. It would be interesting to see which one resolves the issue.

I'll look into it tomorrow.

thomasinaz41213, 2024-03-16:
The steps were as follows...
Change the branch/tag to a version number, setuphelper downloads the version, I think setuphelper has no way of knowing there will be an incomplete fileset because the offending file is a .patch file (cannot complete patch), so that part I think is to be expected.
Next I press install, and then get the incomplete fileset. Back out of the package and then select the package again, or any other package, and get the setuphelper screen. Next/previous buttons do not change the screen, except that it does seem to scroll through the appropriate number of screens according to how many packages are installed. (Evidenced by the next or previous button becoming shaded according to the number of times pressed)
Restarting just the gui does not correct the problem. Restarting setuphelper does correct the problem.
If I can do anything else let me know.

kwindrem, 2024-03-16:
fixed in SetupHelper v7.1

Thanks for reporting and help with debugging. It would have taken me a long time to figure it out without the details provided above.

thomasinaz41213, 2024-03-17:
Still seeing this behavior with v7.1. In addition, after manually installing a package from the active packages menu, I get the prompt to reboot now or later. I press 'now' and the screen shows 'reboot pending' but never reboots. I waited about 10 minutes to see but it never reboot. I was able to recreate this on two different systems with different packages being installed.

thomasinaz41213, 2024-03-17:
Found one more issue. After a firmware update, and all active packages are reinstalled, system has to be reboot for guimods gui changes to take effect. Re-created on several Venus os versions.

kwindrem, 2024-03-17:
Are you still seeing Download, Install, Uninstall/Remove grayed out and not able to switch to a different package?

Does your package require reboot?

Can you get to the log for PackageManager, seeing that may help. If so, please zip it and post here.

/data/log/PackagManager/current

If you can run dbus-spy, what are the contents of /GuiEditAction and /GuiEditStaus in com.victronenergy.packageManager?

Also, each package has a set of parameters. Look at these to see if anything stands out. Packages are in the same order as they are in the Active packages menu.

Also see if /ActionRequored

I did simulate this issue and thought I'd fixed it. Let me explain what is supposed to happen:

PackageManager checks one package per second for issues that would result in an install failure and block the install from the Package editor menu if any are found. Besides version and platform checks and compatibility checks I urn the setup script in "check" mode to attempt to create missing file sets and (as of v7.1) also patch all files to a temporary location to see if any patch scripts fail. These checks would block or enable installs (manual from the Pacage editor menu or automatically if Auto Install is turned on). Any issues detected in this way persist until they are resolved.

If install is triggered, these same checks are made again and PackageManager gets the failure return code from calling the setup script. These failures are reported to the GUI with an OK button but once dismissed they aren't show again. But any detected by the scan described above will continue to be displayed.

There are some descrepencies in showing "reboot now ?" and what happens when you press Proceed/Now so ignore that one for now.

kwindrem, 2024-03-17:
On the issue that GuiMods is requisting a reboot:

The need to reboot may be coming from another package install/uninstall. PackageManager collects the GUI restart and reboot requests for all operations and forma s single action needed indication for the GUI. This controls what is shown when any operation completes. So if a package requires reboot and you dismiss that with Later, then install GuiMods, the reboot needed message will reappear even though it was the other package that requires reboot. Could this be what you are seeing?

thomasinaz41213, 2024-03-17:
On the last issue that I reported about the gui needing reboot after a firmware update...this is after setuphelper automatically installs the active packages to the new firmware (no manual interactions). If I navigate to the active packages menu, all the packages show as being installed with no messages in the packages listed, no prompt or message that a reboot is needed. (Gui did reboot a few times during the time that setuphelper was installing the packages). Everything appears as normal in the package manager menus, but the gui is still on victron oem. I left it sit for an hour just in case it was busy doing something else but it never self corrected.

I will do some further testing to answer your other questions.

One thing to note (which I will check and report back), is that I cannot recall if I actually tried tapping 'OK' when prompted after the failed install attempt. But even if you don't tap 'OK' should it be stuck in a loop like that?

thomasinaz41213, 2024-03-17:
If I manually install a package (any package, doesn't matter which one), then I am prompted for a reboot now or later, select now and then it says reboot pending. Guieditaction shows reboot and guieditstatus shows reboot pending. It stays in that state indefinitely or until I manually reboot. After reboot I attempted to remove an uninstalled package from the active package list but am not able to. Tapping on remove does nothing. I have to revert to an older version to remove packages.

I removed everything except setuphelper and guimods, then updated back to v7.1. uninstalled guimods since it does not require reboot, only gui restart, and get the message 'gui restart pending'. it hangs there indefinitely unless i manually restart. guieditaction shows restartgui, and guieditstatus shows gui restart pending. Actionneeded shows guirestart. same issue of the package page showing setuphelper no matter if you scroll next/previous. you mentioned 'actionrequired' above but the dbus lists it as 'actionneeded'. just a typo above? or is this the problem?

thomasinaz41213, 2024-03-17:
fyi the action CHECK is causing thousands of entrys into the package log.

kwindrem, 2024-03-18:
Sorry for the delay but this one took me a while to sort out. v7.2 is available now.

The reporting of reboot and gui restart needed is a bit misleading in that is shows what is needed for all installs and uninstalls, not just what is needed for the package shown in the Package editor menu. If you don't think this can be understood, I can go back to an action needed for each package. There is also a global indicator in the Package manager main menu. Another unexpected thing is that switching to a different package will bring the reboot/restrt needed message and Now / Later buttons back.

The Package editor menu had some issues with lockup if you attempted to navigate away from the current package while something was pending (OK or Confirm/Canecel ). Now navigating will cancel the pending operation.

thomasinaz41213, 2024-03-19:
All of the previous issues seem to be resolved, but there is another that has popped up concerning conflicting packages. If I click on a package that has a conflict and then click to resolve the issue I get an error.

[image]

[image]

[image]

kwindrem, 2024-03-19:
OK. I'll get on that. Thanks for the prompt testing.

thomasinaz41213, 2024-03-19:
Still cannot remove packages.

kwindrem, 2024-03-19:
fixed: GitHub versions are not refreshed when reentering the Active Packages menu
fixed: unrecognized command showDetails
fixed: can't remove packages from Package editor menu
improve reporting of restart/reboot requirements in GUI

The restart/reboot status now lists all required package/operaton and what will happen when you press Now

drtinaz, 2024-03-19:
Confirmed I am now able to remove packages. However I now see a duplicate package in the list? They both show the same GitHub username and branch. I did just change my GitHub username so maybe that has something to do with it, but I don't see double entries for ContextStorage which is also my repo. What are your thoughts? Just remove them and re-add and forget about it?

drtinaz, 2024-03-19:
More on the issue of duplicates in the active packages list:
This seems to happen when removing a package from the active packages list. After tapping on proceed (to remove), the package desired to be removed is no longer listed, but then one of the other packages in the list is duplicated. Re-initializing package manager seems to be the only resolution.

kwindrem, 2024-03-19:
SetupHelper v7.7 is out now that fixes the duplicate in the package list caused by a remove package.

PackageManger now checks for duplicates in the list during initialization and removes them from the list.

It also checks for duplicates during normal operation and triggers a PackageManager restart which will then remove the duplicates. This should help keep the package list error free.

Thanks again for detailed testing. You have really helped make SetupHelper better and more solid.

closed 2024-03-25 by drtinaz