kwindrem/SetupHelper #79: Package update reboot

drtinaz, 2024-03-21:
Not sure if this is an issue or my failure to understand. When pushing a new release for a package, what is the sequence of events that takes place in setuphelper? Reason I ask is if I push a new release that has an update to the driver in the package, a reboot is not occuring. I have to manually reboot, or stop/restart the driver. The package is leveraging the auto install files resources. Should this occur automatically, or do I need to add a reboot required to the setup script?

drtinaz, 2024-03-21:
I would think any package that installs/runs a service should at the very least restart the service when the package updates?

kwindrem, 2024-03-21:
Yes, that is true. It does. 

However, there is no way for the service to tell PackageManager a reboot is needed. Services can generally just be restarted svc -t /service/< service name >

It would be possible to include a REBOOT flag file in the service directory that the setup script could then look for and deal with the reboot in the same way reboots for files are handled.

drtinaz, 2024-03-21:
Can I just add svc -t /service/< service name > before the end script? Or would reboot required be better? What would you suggest?

kwindrem, 2024-03-21:
ServiceResources should already be restarting the service if is is running. You should see a log entry if this happens.

A newly installed service should start on its own and I do not attempt to start it.

If this is not happening, let me know and I'll check my code

I would only request a reboot if there is something that is only checked or used in the boot process (like udev rules or gpio_list or config.txt changes )

kwindrem, 2024-03-21:
PackageManager does not check modification dates so won't see your changes.

If you run the package setup script it will restart your driver service:

drtinaz, 2024-03-21:
I was pushing releases thru GitHub.

kwindrem, 2024-03-21:
Should work if you have auto downloads and auto installs turned on AND you change the version number.

If the version number in the branch/tag at GitHub doesn't change then PackageManager won't know there's anything to do.

You can trick PackageManager to download by removing the version file in the package directory.

You can trick PackagerManager to install by removing /etc/venus/installedVersion-< package name >

closed 2024-03-25 by drtinaz