kwindrem/SetupHelper #81: updatePackage giving errors

drtinaz, 2024-03-23:
I pulled the updated SetupHelper to my developement PC (was previous v7.0), and now when I run updatePackage "packagenamehere" I get the following:

./updatePackage: line 2: $'\r': command not found
./updatePackage: line 38: $'\r': command not found
./updatePackage: line 89: $'\r': command not found
./updatePackage: line 93: $'\r': command not found
./updatePackage: line 94: $'\r': command not found
./updatePackage: line 101: $'\r': command not found
./updatePackage: line 105: $'\r': command not found
./updatePackage: line 109: $'\r': command not found
./updatePackage: line 113: syntax error near unexpected token `elif'
'/updatePackage: line 113: `elif [ ! -e "$stockFiles" ]; then

kwindrem, 2024-03-23:
All except the last of these are blank lines and the \r suggests an end of line issue in converting from unix text files to Windows.

I and not all that familiar with what is supposed to happen during the transfer. I'm wondering that since the file doesn't have an extension maybe Windows can't deal with it. 

If adding .sh at the end (updatePackage.sh) helps let me know.

drtinaz, 2024-03-23:
You know now that you mentioned eol, I forgot that I needed to change from CRLF to LF last time I loaded it onto my development PC. Man that gets me every time.

kwindrem, 2024-03-23:
Glad that was it.

closed 2024-03-23 by kwindrem