kwindrem/SetupHelper #84: Incomplete file set for v3.40~2

wetty, 2024-04-12:
Hello,

i tried to install SetupHelper on my RaspI Venus OS V3.40 2 and get the following error

root@venuspi:~# /data/SetupHelper/setup
--- starting setup script v4.23
creating package options directory /data/setupOptions/SetupHelper
creating file set for v3.40~2
ERROR v3.40~2 PageSettings.qml no replacement file
ERROR v3.40~2 PageSettingsPackageManager.qml no replacement file
ERROR: incomplete file set for v3.40~2 - can't continue

kwindrem, 2024-04-12:
Not sure where you got SetupHelper v4.23 but it's very obsolete. The latest is v7.15. Get a new copy from the latest tag in GitHub or use the blind install you find there.

closed 2024-04-12 by kwindrem

wetty, 2024-04-12:
I did get the latest Version with the provided wget command
root@venuspi:/data# wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C /data

It might be possible that I tried SetupHelper a year ago and that the files where not all correctly removed. And so the new ones could not copied over them. I did delete two SetupHelper Folders, downloaded all again and installed it. Now its working.
Thanks for the help.