kwindrem/SetupHelper #85: Problem removing services when uninstalling

drtinaz, 2024-04-12:
I have a package with 2 services in the 'services' folder. Package installs fine. When uninstalling gets to 'removing services' portion the first service is removed but the following errors are popped

ServiceResources: line 119: [: ==: unary operator expected

ServiceResources: line 121: [: ==: unary operator expected

and the second service then has to be manually removed.

Venus v3.30 on a cerbo gx
Setuphelper v7.16

kwindrem, 2024-04-12:
The problem is with a long name. Fixing the issue now. v7.17 out shortly

drtinaz, 2024-04-13:
That fixed the errors, but only one service is being uninstalled. There are two services in my package, both are correctly installed, but only one gets uninstalled.

kwindrem, 2024-04-13:
You are correct, and I just fixed it. V7.18 is out now.

closed 2024-04-13 by kwindrem