kwindrem/SetupHelper #89: Version 8 high data usage

lfhoward, 2024-05-16:
Hi Kevin,

Version 8 of SetupHelper just came out last night, which is awesome. Unfortunately my data usage went from approximately 0.5 MB per hour to about 5.5 MB per hour as soon as the new version was auto-installed. See the pic below of data usage from my SixFab, my cellular carrier:

[image]

I have a mobile installation using a Raspberry Pi 4 as a cellular modem, which costs $9 a month if I keep data usage below 500 MB per month. Up to now that has worked. At the current data usage rate that is 11x higher, I am looking at $58-$87 per month for 4-6 GB of data... I would need to take my installation offline or come up with a different mobile internet solution. 

Would you be able to take a look at the data usage rate in version 8 of SetupHelper and see if you can bring it back to or close to pre-update levels?

Thanks for all you do for us in the Victron community! 
Sincerely,
Lauren

kwindrem, 2024-05-16:
I will certainly look into this. It should not be happening. 

Until then, if you have not done so already, turn off automatic downloads.

lfhoward, 2024-05-16:
Hi Kevin,Thanks a lot for taking a look at the code! I toggled Automatic GitHub Downloads to OFF for now. Best,Lauren

kwindrem, 2024-05-16:
I found a bug that was requesting version information about every 10 seconds rather than every 10 minutes. This should make a significant difference. v8.1 is out now and will download with the branch/tag set to latest.

I am considering an enhancement to set the GitHub version refresh to once per day. But let me know if you think that's necessary after you log some data usage with 8.1 installed.

Thanks for reporting this. Most of us would not notice.

lfhoward, 2024-05-17:
Kevin,Excellent. I’ll download 8.1 and let you know how it goes. Thank you for being so responsive. Lauren

lfhoward, 2024-05-17:
Data usage is back to what it was before. I would be happy to have SetupHelper check once an hour or once every 6 hours. This would free up more data to be able to download OS updates and stay under my cap of 500 MB per month. 

[image]

kwindrem, 2024-05-17:
I just checked in SetupHelper v8.3 that adds GitHub check frequency: 10 minutes (the rate previously called Normal), Hourly and Daily. I'd be interested in how much this affects your network usage.

FYI, the version checks are spaced evenly across the specified time. For example, for every 10 minutes and 8 packages in the list that's one version check every 75 seconds. With 4 packages and Daily, that's one version check every 6 hours.

All GitHub versions are still refreshed quickly when entering the Active Packages menu. One package refresh occurs when navigating packages in the Package edit menu.

lfhoward, 2024-05-18:
Following up on data usage:

Version 8.1 brought the data usage back in line to what it was before v. 8.0. Version 8.2 cranked the data rate back up to v. 8.0 levels. I just downloaded 8.3 and set the interval to “daily” so I’ll let you know what happens. The spike on the most recent hour is probably me downloading 8.3 on top of the high data usage from 8.2.

[image]

lfhoward, 2024-05-19:
Results of setting the refresh rate to daily:

[image]

Minimal data usage!!! This makes me happy. It is even less than before, so that means I have more leeway to play around with the system, download updates, and not have to worry about hitting my monthly data cap. 

Thank you Kevin!

lfhoward, 2024-05-20:
Data usage continues to be low and manageable.  This flexibility is a really good feature of the new version. 

[image]

kwindrem, 2024-05-20:
Glad it's working well for you.

Closing issue now

closed 2024-05-20 by kwindrem