kwindrem/SetupHelper #90: Log handler error

drtinaz, 2024-05-23:
Setup helper was on version 8.0.
Tried to uninstall a package that was installed back when setuphelper version was still in the 7.xx range and got an error log handler not found. I reverted back to setup helper version 7.10 and was able to uninstall the package. Looks like a backward compatibility issue?

kwindrem, 2024-05-24:
This is definitely something I want to look into.

I'd need more details to look into this. Which package had the problem and what version was it.

Logs would also help. SetupHelper's ReadMe has instructions for accessing logs without ssh capability.

Also note, I'm away from my development system for a few days.

drtinaz, 2024-05-24:
I'll be traveling the next several days as well. I have a few more details for you in the meantime...
When I was unable to uninstall my package, I reverted to v7.10 of setup helper and was able to uninstall my package successfully. Then I upgraded to v8.3 of setup helper and tried to install another package I just finished up, but got the same error. Then I remembered that there have been some major changes between 7.xx and 8.xx so I went searching in the setup helper package. Sure enough, the file path references in the setup script of (choose any package) have changed between versions. I updated the file references to the helper files inside the setup script and install/uninstall functions properly again. 

Most people probably wouldn't run into this issue, but anyone running an older package with the latest package manager would need to update their setup script or run into the same error.

kwindrem, 2024-05-27:
I see the problem. Working on a fix. Thanks for reporting the issue.

kwindrem, 2024-05-27:
Should be fixed now in SetupHelper v8.4

closed 2024-05-27 by kwindrem