kwindrem/SetupHelper #92: download pending

wilma75, 2024-07-02:
Hello,
unfortunately I have a problem. I would like to install the ShutdownManager. But it just says download pending and nothing happens. Even after waiting a long time.
The same thing happens with other packages

kwindrem, 2024-07-02:
Have you tried rebooting?

What firmware version?

What SetupHelper version?

Logs would help (see SetupHelper ReadMe)

wilma75, 2024-07-02:
Hello, oh damn sorry.
So Venus OS is V3.34, Setuphelper is 8.5, everything is new.
I rebooted several times.
I installed it on the raspy using a USB stick. It didn't work via SSH either. Maybe it's stuck somewhere.

kwindrem, 2024-07-03:
There's an error message in the log for Package Manager. I'll look into it.