kwindrem/SetupHelper #93: Unable to install on V3.33 "Package manager not running"

TheRealBaldingEagle, 2024-07-04:
I had setup helper and GUI mods working perfectly on V2.93 but I added the Orion XS and wanted to update to newer version. Now I am unable to install setup helper. I get the message "package manager not running" I have tried multiple times to install setup helper, I tried reverting back to V2.93 and all with no luck. Thanks for the help!

kwindrem, 2024-07-04:
What version of SetupHelper were you running?

I assume you have tried rebooting.

It will be hard to diagnose without logs and since PackageManager isn't running you will need ssh access to gather the logs.

You might try the SetupHelper blind install process detailed in the ReadMe file included with the package and on GitHub also. There is also a blind UNINSTALL process that may be necessary. Other troubleshooting help is included there also.

TheRealBaldingEagle, 2024-07-04:
I have tried rebooting many times. I am trying to install V8.0 It may be the USB flash drive I was trying to use. Can you install with an SD card? I was trying to do the blind install.

kwindrem, 2024-07-04:
Any removable media should work (USB stick or SD card).

SetupHelper is currently at v8.6. This version fixed a bug that caused PackageManager to hang. I do not know if that is your problem but would definitely work with the latest version because of that.

I am successfully running SetupHelper v8.6 on v3.33 firmware and just tested the blind install on those versions.

TheRealBaldingEagle, 2024-07-04:
I'm honestly not sure what it is doing now. I ran the blind uninstall and suddenly I'm seeing the GUI mods overlay but package manager is not running. [image][image]

kwindrem, 2024-07-04:
Do you have ssh (terminal) access ? Without the command line interface logs won't be accessible so I can't determine what might be causing PackageManager to not run.

The blind uninstall process might be your only next step.

I hate to have you use the Victron restore to defaults mechanism as that will wipe out all your configuration settings. Plus I'm not sure that will fix PackageManager.

TheRealBaldingEagle, 2024-07-04:
I don't currently have ssh (terminal) access setup. Do I need to plug in with a ethernet cable to my computer? If so, I don't have the hardware to do that at the moment. 

I tried to run a blind uninstall already I think. I downloaded the venus-data-UninstallPackages.tgz and changed the name and rebooted the cerbo as instructed.

kwindrem, 2024-07-04:
One other thing to try would be to reload the firmware. Do this by going to stored backup firmware and selecting the previous version to reboot to that version. Then go to online update and install a fresh copy from there. That will force SetupHelper to reinstall.

These are all shot in the dark ideas since the only way to really know what the problem is would be to look at the logs.

Cerbo can have its own WiFi access point which you can connect to from your laptop. Then you can go into Settings / General and setup ssh access. You need to enable root access and on Cerbo that is accomplished by dragging down on the access level and holding at the bottom of the screen for 5 seconds. You will need a root password.

You can then use putty or terminal in Mac OS to access the Cerbo's CLI. If you get that far you can access the log with:

`tail -50 /data/log/PackageManager/current | tai64nlocal
`

Fgonzalez22, 2024-07-06:
Hi! im having the same issue. Package manager not runnning.
Currently I use Firmware 3.30 on a Raspberry pi 3b. 
I don´t know when package manager stopped working.
I have tried updating firmware to 3.34 Large but I loose configuration of tanks, relays, gui and i still dont have package manager running.
went back to 3.30 and got back settings. Packages are forking. Only the package manager is not running.

Tried restoring setup manager with shh. I get the following error:
data/SetupHelper/setup: line16: data/SetupHelper/HelperResources/IncludeHelpers: No such file or directory

after: tail -50 /data/log/PackageManager/current | tai64nlocal 

i get 50 lines with almost the same timestamp  and: ./run: line3: data/SetupHelper/PackageManager.py: no such file or directory

Thanks for any help!

kwindrem, 2024-07-06:
Very strange. line 16 in IncludeHelpers for the latest SetupHelper version is getting the full path to the directory where the setup script exists. You could check that file with nano to see if it's the same line I'm seeing. Or let me know what SetupHelper version is currently loaded on the system: cat /data/SetupHelper/version

The error from the run file indicates that SetupHelper setup probably did not complete and the run file is left over from a previous install prior to moving services around in the Package. That's consistent with the first error.

I would suggest running a SetupHelper blind install ... or getting a fresh copy of SetupHelper from GitHub (explained in the SetupHelper ReadMe) then running the setup script again from the command line.

Fgonzalez22, 2024-07-06:
I have no  programing knowledge, but after running the command line you suggested i get  im running v10.29

I have installed setuphelper from the command line the first time. If running from a Raspi 3b, can i do a blind install from a usb memory stick? Will I lose my settings?

kwindrem, 2024-07-06:
v10.29 sounds like a GuiMods version. SetupHelper is currently v8.6.

Yes you can do the SetupHelper blind install using a USB memory stick. It will not disturb your settings. Make sure you use the venus-data.tgz file from the SetupHelper package and not from the Victron restore to factory defaults mechanism (the file names are the same).

Fgonzalez22, 2024-07-06:
I  will try next time I go to the boat, now I'm via teamviewer to openplotter and from there to the venus raspi with shh.
I could run nano and i got the image attached
[image]

Fgonzalez22, 2024-07-15:
Blind install successfully restored package manager and did not lose settings. Thanks a lot!