kwindrem/SetupHelper #96: Service run files executable permissions lost when SetupHelper copies package from removable media

Seneda, 2024-08-14:
I have been working on a package to run on an Ekrano GX unit. The package is a python script which will just run as a service and I am using the removable media option to install it.

The package/services folder contains a `run` file which I have made executable (`chmod +x`) , but for some reason when the package is automatically transferred from the removable media to `/data/packagename` the run file has lost executable permissions and the script fails to start.

But if i ssh into the system and `chmod +x` the file then the service starts and works as expected.

Also if I manually un-tar (`tar -zxf`) the package then the run files do have their executable property. Is there anything in the process that transfers the files from the .tar.gz file to the `/data/packagename` directory that would modify these properties?

```
root@ekrano:/media/sda1# tar -xzf EVP15KVAGeneratorControl-0.1.10.tar.gz
root@ekrano:/media/sda1# ls -l EVP15KVAGeneratorControl-0.1.10/services/generator_control/
drwxr-xr-x    2 root     root          4096 Aug 14 13:53 log
-rwxr-xr-x    1 root     root           117 Aug 14 13:53 run
root@ekrano:/media/sda1# ls -l EVP15KVAGeneratorControl-0.1.10/services/generator_control/log/
-rwxr-xr-x    1 root     root            72 Aug 14 13:53 run
root@ekrano:/media/sda1# ls -l /data/EVP15KVAGeneratorControl/services/
drwxrwxr-x    3 root     root          4096 Aug 14 13:53 generator_control
root@ekrano:/media/sda1# ls -l /data/EVP15KVAGeneratorControl/services/generator_control/
drwxrwxr-x    2 root     root          4096 Aug 14 13:53 log
-rw-rw-r--    1 root     root           117 Aug 14 13:53 run
root@ekrano:/media/sda1# ls -l /data/EVP15KVAGeneratorControl/services/generator_control/log/
-rw-rw-r--    1 root     root            72 Aug 14 13:53 run
```

Seneda, 2024-08-14:
There is an argument to tar `-p, --preserve-permissions, --same-permissions` which will supposedly preserve permissions, but presumably this already normally works for everyone else, so i'm guessing something is unusual in my system so this may not be relevant.

kwindrem, 2024-08-14:
I have not experienced the problem you are seeing.

The code in PackageManager.py does not specifically set the preserve option and the copy is done with a python shutl.move () call.

Code in ServiceResources is responsible for copying the service directory including the run files from the package directory to /service and /opt/victronenergy/service. This code does not change file permissions either.

Which copy of the file is loosing the executable permission: in /data/<package name>/services/<service name> or in /service/<service name> ?

Seneda, 2024-08-14:
The copy in /data//services is losing the executable permission. 
However strangely the setup script which also has that permission set does not lose it.
I suppose I could add something to my setup script to modify the file permissions of the run files in /data///services, but id be a bit uncomfortable to do that without figuring out the cause.

Seneda, 2024-08-14:
Apologies for wasting your time, but it seems that the commit in which i thought i set the properties did not actually have that change in it. And for whatever reason when i run tar -xzf in the ssh session it was adding the executable permission even though it was not set on those files, which made me believe that the contents of the tar file did indeed have the permission set. In short i have added the executable permission again explicitly through git  `git update-index --chmod=+x path/to/file` and all seems to work as expected.

closed 2024-08-14 by Seneda