kwindrem/SetupHelper #97: WARNING:RemoteGPIO install failed - package conflict cat: can't open '/data/SetupHelper/HelperResources/version': No such file or directory

Nvanderbeek, 2024-09-08:
I have some issues with a clean install on my setup.

It is a CM4 configuration with waveshare CM4-IO-WIRELESS-BASE-B  Board

On installing the setuphelper script and installing packages the log complains on RpiGpioSetup on missing directory on installation

@4000000066dd759702cc1e2c WARNING:RemoteGPIO install failed - package conflict cat: can't open '/data/SetupHelper/HelperResources/version': No such file or directory
@4000000066dd759702cc3984 mkdir: can't create directory '/data/RemoteGPIO/conf': File exists
[logs.zip](https://github.com/user-attachments/files/16921892/logs.zip)

what is the issue?

kwindrem, 2024-09-08:
The errors are coming from the RemoteGPIO setup script which is maintained by a different author. You might need to contact them for assistance.

RpiGpioSetup and RemoteGPIO are mutually exclusive and there should be code to prevent one from installing if the other is already installed but this is apparently no longer working for the RemoteGPIO install. I had previously tested this previously and seemed to work at the time.

I did make some recent changes to SetupHelper. Prior to these changes, the helper resources could come from within the package. This has since been eliminated and a change to the helper resources include in the package setup script has changed. This may be causing RemoteGPIO setup scrip to use outdated helper resources. This is most certainly the reason for the HelperResources/version error.

kwindrem, 2024-09-08:
I have updated SetupHelper to eliminate the HelperResources/version error.

This also permits the conflict checking to prevent RemoteGPIO install if RpiGpioSetup is already installed. I believe this check was failing previously since the setup script was crashing prior to performing the conflict checks.

closed 2024-09-08 by kwindrem

Lucifer06, 2024-09-09:
Thx Kevin for the heads up and for fixing the issue from your side.
I will soon, update RemoteGPIO to deal with latest SetupHelper resources.