kwindrem/SetupHelper #99: SetupHelper v4.29 to v8.20 PageSettings.qml patch unsuccesful

JimS-JJ, 2024-09-28:
Hi, I've been trying to reinstall SetupHelper with no success.  By SSH, I run the first script
wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar - xzf - -C /data
It works ok. Then I load the 2nd script:
mv /data/SetupHelper-latest /data/SetupHelper
 and I get this message back:
mv: can't rename '/data/SetupHelper-latest': Directory not empty
Then by the 3rd script:
/data/SetupHelper/setup      This message shows up, so I can't uninstall and try again.
--- starting setup script v8.20 action: NONE
PageSettings.qml patch unsuccesful
I have a CerboGX that was running v2.92 upgraded to v3.42 and SetupHelper that was running v4.29 and upgraded to v8.20
I've tried rebooting several times, also the package manager says it is not running, and I can't go into each package and uninstall there.
I even tried removing /etc/venus/installedVersion-SetupHelper Per one of the suggestions I saw to trick PackagerManager to install, but that did not work either.  I am out of ideas.  Can you help?
Thanks for everything you do.  You are very much appreciated. Jim

kwindrem, 2024-09-28:
There have been reports of upgrades from much older SetupHelper versions failing as you are seeing. In these cases, I recommend uninstalling all packages, upgrading SetupHelper then upgrading and reinstalling all packages from PackageManager.

From the text above, it appears you already have SetupHelper v8.20 (the latest) installed. 
To get around the error you are seeing, specify uninstall on the command line to skip the checks resulting in the error you are seeing. Eg:

/data/SetupHelper/setup uninstall deferReboot deferGuiRestart
/data/GuiMods/setup uninstall deferReboot deferGuiRestart
etc
Then install SetupHelper from the command line:
/data/SetupHelper/setup install deferReboot deferGuiRestart
If you are running on a PI, install RpiDisplaySetup from the command line so you'll have the GUI on the local display ager rebooting:
/data/RpiDisplaySetup reinstall
Otherwise, reboot

PackageManager should now be running and you can download the latest version of your packages and install them.

The directory not empty error is probably due to an existing /data/SetupHelper directory. Try:
mv -f /data/SetupHelper-latest /data/SetupHelper
This will overwrite an existing target instead of preserving it and aborting the move.

What is most likely happening in this case is that any attempt to run /data/SetupHelper/setup 

The second error is most likely due to running the older SetupHelper version.

Removing /etc/venus/installedVersion-SetupHelper tells PackageManager (and the setup script for the package if running that from the command line) that the package is not yet installed. But since PackageManager is not running it can't do the install. A reboot MIGHT work but if you still have the v4.29 SetupHelper in place that would not get you past the error.

BTW, the recommended install method is via blind install, but in your case it probably won't get you around the issues you are having. Blind install is explained in the SetupHelper ReadMe.

JimS-JJ, 2024-09-28:
Wow, that fixed me.  I was almost ready to start over assuming I had bricked my device, but then I thought well, everything is still sort of operating, so I thank you very much for getting me out of a this bind.  I'm sure this will help others too.  Couldn't have done it without you. I am just learning some linux commands, but not enough to know how it all works yet.  Thanks again.

closed 2024-09-28 by kwindrem